eclipse-zenoh/zenoh-plugin-ros2dds #12: [Bug] Unable to communicate while using ROS_LOCALHOST_ONLY

evshary, 2023-10-16:
### Describe the bug

While using CycloneDDS + ROS_LOCALHOST_ONLY + Disable multicast on localhost, talker & listener can talk to each other.

### To reproduce

1. Terminal 1
```shell
ROS_DOMAIN_ID=1 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ROS_LOCALHOST_ONLY=1 ros2 run demo_nodes_cpp talker
```
2. Terminal 2
```shell
ROS_DOMAIN_ID=2 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ROS_LOCALHOST_ONLY=1 ros2 run demo_nodes_cpp listener
```
3. Terminal 3
```shell
./target/release/zenoh-bridge-ros2dds -d 1 --ros-localhost-only
```
4. Terminal 4
```shell
./target/release/zenoh-bridge-ros2dds -d 2 --ros-localhost-only
```
5. Talker and listener can't talk to each other, but we can enable multicast to fix this.
```shell
sudo ip link set lo multicast on
```
6. Also, if we use FastDDS, then this issue also disappears.

## Extra information

I also tried the same thing on zenoh-bridge-dds, and found that the issue also appeared in 0.10.0 but disappeared in 0.7.0.
Therefore, I guess this is related to Zenoh version.
Here is my environment for zenoh-bridge-dds: https://github.com/evshary/zenoh_demo_docker_env/tree/main/ros_with_zenoh_bridge_dds

### System info

Platform: Ubuntu 22.04
ROS version: Humble
Bridge version: main branch

JEnoch, 2023-11-03:
In Humble and previous version, `ROS_LOCALHOST_ONLY=1` makes `rmw_cyclonedds_cpp` to configure CycloneDDS with:
```xml
<CycloneDDS><Domain><General><Interfaces>
   <NetworkInterface address="127.0.0.1"/>
</Interfaces></General></Domain></CycloneDDS>
```
With this config, CycloneDDS behaves differently, depending if the `127.0.0.1` interface has multicast enabled or not:
- is enabled, multicast over loopback is used
- if disabled, CycloneDDS tries to contact a set of Participants via unicast on `127.0.0.1` on different expected port numbers. The number of tried ports depends on config `<CycloneDDS><Domain><Discovery><MaxAutoParticipantIndex>` which is **9** by default.

Now, both `zenoh-bridge-dds` and `zenoh-bridge-ros2dds`, since version `0.7.2-rc` and https://github.com/eclipse-zenoh/zenoh-plugin-dds/pull/114 use this configuration for CycloneDDS:
```xml
<CycloneDDS><Domain><General><Interfaces>
   <NetworkInterface address="127.0.0.1" multicast="true"/>
</Interfaces></General></Domain></CycloneDDS>
```
Meaning that the usage of multicast over loopback is forced, even if the interface declares it's not enabled.
Therefore, it's no longer compatible with CycloneDDS default behaviour when multicast is disabled on loopback.

While enabling multicast on `lo` interface at host startup is the best option as explained [here](https://github.com/eclipse-cyclonedds/cyclonedds/issues/1400#issuecomment-1241575011), it was probably an error to force the bridges to use multicast over loopback, if `rmw_cyclonedds_cpp` doesn't do the same.
I will remove this setting from the 2 bridges (dds & ros2dds).

marredir, 2023-11-06:
Hi @JEnoch, thanks for the info: probably just partially related to this ticket, but could you guys please release some docs/whatever about the demos you had with the turtlebot4 robots at RosCon23 ( I have tried to look for it but I can not find it)? Besides helping specifically for that platform I think that is a very good "tutorial" as it has a slightly different configuration (create3 <- USB -> raspberry <- Wifi -> Workstation). I have found a very brief mention [here](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds#usage) but some more details (if needed) would help extremely helpful. Thanks

JEnoch, 2023-11-06:
> could you guys please release some docs/whatever about the demos you had with the turtlebot4 robots at RosCon23

Hi @marredir, yes, that's coming very soon, as well as an online webinar in December 🙂

commit bb39e219 (2023-11-06): When ROS_LOCALHOST_ONLY=1 don't force multicast on lo (fix #12) (#15)
closed 2023-11-06 by JEnoch