eclipse-zenoh/zenoh-plugin-ros2dds #13: [Bug] ROS 2 node crashes while querying admin space

evshary, 2023-10-17:
### Describe the bug

While running ROS 2 node, using curl to query admin space will cause the node to crash.

### To reproduce

 1. In terminal 1, run talker
 ```shell
ros2 run demo_nodes_cpp talker
````
2. In terminal 2, run zenoh-bridge-ros2dds
```shell
./target/release/zenoh-bridge-ros2dds --rest-http-port 8000
```
3. Query the admin space
```shell
curl http://127.0.0.1:8000/\*\*
```
4. terminal 1 will crash
```
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
```

### System info

Platform: Ubuntu 22.04
ROS version: Humble
Bridge version: main branch

JEnoch, 2023-11-20:
> `curl http://127.0.0.1:8000/\*\*` 

Such a key expression (`**`)  doesn't query only the admin space, but ALL Zenoh key space, that includes both user space and admin space.
Therefore, ALL the queryables declared by the bridge are queried, including the ones mapping to ROS 2 Services and Actions. But as your query has not body payload, the ROS 2 Service (probably related to ROS parameters in your case) fails to deserialize the payload and crashes.

In general, it's not a good idea to query on `**` as it could trigger some queryables that have some expectation wrt. the query.
The `zenoh-bridge-ros2dds` admin space is on `@ros2/**`. You should query it as such:
```bash
curl http://127.0.0.1:8000/@ros2/\*\*
```

In the bridge, I will add a safety check in queryables for Services and Action avoiding to forward to ROS an empty request.
That would at least prevent crashes by accident of a query without payload.
Still, nothing can prevent a malformed payload to make a ROS Node to crash, except defensive code inside ROS itself.

evshary, 2023-11-20:
That makes sense. I also found this by accident.
We indeed can't prevent wrong requests to ROS Node and crash it.
It's fine for me to at least add the safety check on empty requests, since users might run this mistakenly.