eclipse-zenoh/zenoh-plugin-ros2dds #28: [Bug]

Fellfalla, 2023-12-09:
### Describe the bug

Running the bridge I get following error with the queries timeout arg:

```
2023-12-09T15:47:33Z INFO  zenoh::net::runtime::orchestrator] Zenoh can be reached at: tcp/172.22.0.1:7447
thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: Plugin `ros2dds` configuration error: invalid type: floating point `60`, expected struct QueriesTimeouts at zenoh-plugin-ros2dds/src/lib.rs:130.', zenoh-bridge-ros2dds/src/main.rs:220:66
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
[ros2run]: Aborted
make: *** [Makefile:54: zenoh] Error 250
```

### To reproduce

Run 

```
ros2 run zenoh_bridge_ros2dds zenoh_bridge_ros2dds \
		-l tcp/0.0.0.0:7447 \
		--no-multicast-scouting \
		--queries-timeout 60.0
```

### System info

- Ubuntu 22.04 in docker image based on osrf/ros:humble-desktop-full
- zenoh-plugin-ros2dds hash: eb66be54a0f5af584134843fe86a2fa02493496c

Fellfalla, 2023-12-10:
@JEnoch  I had a look at it and I think something with the argument parsing in `main.rs` line 183 is didn't get updated accordingly with commit 63382f2a3ece424544d5f1cb21f6962ac2f24e30. As far as i understand, config.rs expects a QueriesTimeout struct rather than a simple f64. 

We could apply the timeout as default to all possible values inside the struct:
* for all transient_local_subscribers: `transient_local_subscribers: <my-timeout>`
* for all services: `services: [".*=<my-timeout>"]`
* for all actions `actions: { send_goal: <my-timeout>,  cancel_goal: <my-timeout>,  get_result: [".*=<my-timeout>"] }`

JEnoch, 2023-12-11:
You correctly analysed the issue: I forgot to update the `--queries-timeout` CLI argument following #19 ... 🤦🏻

My only concern with the solution you propose is that by Zenoh convention, all the CLI arguments overwrite the configurations provided by a file (via `-c`). Thus, a single float set with `--queries-timeout` will possibly overwrite a bunch of different timeout settings.
Why not... but I think in such case a warning should be logged.
What do you think ?

Fellfalla, 2023-12-12:
Alternatively we can put the same structure that is going to be parsed as a string into the cli. But for that case users would need tome example on that syntax. 

And Im not familiar enough with the code to quickly implement this 😅.

JEnoch, 2023-12-18:
I finally preferred to explicitly expose a `queries_timeout/default` setting in configuration file that applies if a timeout for a query is not explicitly set via regex in other `queries_timeout` parameters.
And replace the `--queries-timeout` argument with a `--queries-timeout-default` argument that overwrites this `queries_timeout/default` setting.
See #35 for details.

PR #35 (2023-12-18): Add --queries-timeout-default option
Fix #28.

- In configuration file, adds a `default` field to the `queries_timeout` as such:
```json5
      //// queries_timeout: Timeouts configuration for various Zenoh queries.
      ////                  Each field is optional, and a default value of 5.0 seconds applies if not set.
      ////                  Each value can be either a timeout (float in seconds) that will apply to all,
      ////                  either a list of strings with format "<regex>=<float>" where:
      ////                      - "regex" is a regular expression matching an int [... cut]
closed 2023-12-18 by Mallets