eclipse-zenoh/zenoh-plugin-ros2dds #31: [Bug] Issue with nav2 when using the bridge

samiamlabs, 2023-12-15:
### Describe the bug

When the zenoh bridge is running and I launch navigation2, the lifecycle transitions seem to be acting up. Also, and sending navgoals fail.

An example output for the lifecycle thing is:
```
[controller_server-1] [WARN] [1702648966.041332898] [rcl_lifecycle]: No transition matching 1 found for current state inactive
[controller_server-1] [ERROR] [1702648966.041406110] [controller_server]: Unable to start transition 1 from current state inactive: Transition is not registered., at ./src/rcl_lifecycle.c:355
```
As far as I can tell, the nodes are activated despite the error.

However, when the bridge is running in one of my gazebo simulations, bt_navigator dies after I send a navgoal. Sending navgoals works fine when I don't run the bridge.
```
[rviz2-16] [INFO] [1702651216.046634453] [rviz2]: Setting goal pose: Frame:robot2/map, Position(-0.481345, 2.39804, 0), Orientation(0, 0, 0.865052, 0.501682) = Angle: 2.09051
[bt_navigator-12] terminate called after throwing an instance of 'std::runtime_error'
[bt_navigator-12]   what():  Failed to accept new goal
[bt_navigator-12] 
[bt_navigator-12] [INFO] [1702651216.047840482] [robot2.bt_navigator]: Begin navigating from current location (-0.00, 0.00) to (-0.48, 2.40)
[ERROR] [bt_navigator-12]: process has died [pid 11670, exit code -6, cmd '/opt/ros/humble/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator -r __ns:=/robot2 --params-file /tmp/tmpsue952wd -r odom:=odometry/global -r /tf:=tf -r /tf_static:=tf_static'].
```

I tried starting the bridge after I had already started the simulation with nav2 and made sure that sending navgoals worked.
Sending another navgoal after I started the bridge resulted in:
```
[rviz2-16] [INFO] [1702651950.991863971] [rviz2]: Setting goal pose: Frame:robot1/map, Position(-1.25979, 1.01845, 0), Orientation(0, 0, 0.999542, 0.0302465) = Angle: 3.08109
[bt_navigator-12] [INFO] [1702651950.992806330] [robot1.bt_navigator]: Begin navigating from current location (0.00, 0.71) to (-1.26, 1.02)
[bt_navigator-12] terminate called after throwing an instance of 'std::runtime_error'
[bt_navigator-12]   what():  Failed to accept new goal
[bt_navigator-12] 
[planner_server-10] terminate called after throwing an instance of 'std::runtime_error'
[planner_server-10]   what():  Failed to accept new goal
[planner_server-10] 
[ERROR] [bt_navigator-12]: process has died [pid 22391, exit code -6, cmd '/opt/ros/humble/lib/nav2_bt_navigator/bt_navigator --ros-args -r __node:=bt_navigator -r __ns:=/leo1 --params-file /tmp/tmp1k_whwny -r odom:=odometry/global -r /tf:=tf -r /tf_static:=tf_static'].
[ERROR] [planner_server-10]: process has died [pid 22386, exit code -6, cmd '/opt/ros/humble/lib/nav2_planner/planner_server --ros-args -r __node:=planner_server -r __ns:=/leo1 --params-file /tmp/tmp3be7p5hf -r /tf:=tf -r /tf_static:=tf_static'].
```

### To reproduce

To reproduce the lifecycle errors:
1. Clone https://github.com/ros-planning/navigation2
2. Open navigation2 as a Dev Container
3. export RMW_IMPLEMENTATION=rmw_cyclonedds_cpp in the teminals you use in the container
4. Download the latest `zenoh-bridge-ros2dds` from https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/actions
5. Run `./zenoh-bridge-ros2dds -l tcp/0.0.0.0:7447` in the Dev Container
6. ros2 launch nav2_bringup navigation_launch.py

I have only seen the crashing nodes when I send a navgoal in my own simulation, but I suspect the issue is general.
If you don't have a simulated robot environment to test this in, I can set up a Dev Container with turtlebots or something.

### System info

- Platform: Ubuntu 22.04 in docker
- Processor: AMD Ryzen 5 PRO 7530U

btertoolen, 2023-12-18:
I am facing the exact same issue, running the nav2 controller-server with some custom (but simple) controller plugins. It seems like the lifecycle transitions are attempted twice when zenoh is running. I tested running a simple python node with some prints in the callbacks, and there I saw that its service calls weren't being executed twice. I don't know what would make the lifecycle service calls behave differently. 

When attempting to send the action goal to the controller server, I get the same runtime exception:
```
[controller_server-1] [INFO] [1702891959.734452727] [controller_server]: Received a goal, begin computing control effort.
[controller_server-1] terminate called after throwing an instance of 'std::runtime_error'
[controller_server-1]   what():  Failed to accept new goal
[controller_server-1] 
[ERROR] [controller_server-1]: process has died [pid 1760, exit code -6, cmd '/opt/ros/humble/lib/nav2_controller/controller_server --ros-args -r __node:=controller_server -r __ns:=/ --params-file ***.yaml -r /cmd_vel:=/robot/cmd_vel'].
```

I have been looking into this for some time now, but I still have no idea what could be causing this.

I am also using cyclone-dds, on an Nvida Jetson running Ubuntu 20.04.5 LTS (GNU/Linux 5.10.104-tegra aarch64)

@samiamlabs did you manage to get any new insights?

miltzhaw, 2023-12-20:
I am facing the same issue in a slightly different setting. 
I have a turtlebot2 with nav2 and the Zenoh ro2dds bridge connected over a Zenoh router to another Zenoh ro2dds bridge where I run RVIZ. Both machines use cyclone-dds and DDS traffic remains locally. When I give a goal pose (action) from Rviz the navigation node dies on the turtlebot:
```
[component_container_isolated-8] [INFO] [1703082979.885569222] [bt_navigator]: Begin navigating from current location (-14.87, -0.51) to (-14.94, -0.22)
[component_container_isolated-8] terminate called after throwing an instance of 'std::runtime_error'
[component_container_isolated-8]   what():  Failed to accept new goal
[component_container_isolated-8] 
[component_container_isolated-8] Magick: abort due to signal 6 (SIGABRT) "Abort"...
[ERROR] [component_container_isolated-8]: process has died [pid 1691, exit code -6, cmd '/opt/ros/humble/lib/rclcpp_components/component_container_isolated --ros-args --log-level info --ros-args -r __node:=nav2_container --params-file /tmp/launch_params_9hiirbx0 --params-file /tmp/launch_params_aox_pcps -r /tf:=tf -r /tf_static:=tf_static'].
```
The Zenoh bridge gives a warning when this happens which is as follows:

```
[2023-12-20T14:20:36Z WARN  zenoh_plugin_ros2dds::route_service_srv] Route Service Server (ROS:/compute_path_to_pose/_action/send_goal <-> Zenoh:tb2-1/compute_path_to_pose/_action/send_goal): received response for another client: 70f786a63f39626e (me: a9dc1c8a3b36783)
[2023-12-20T14:20:36Z WARN  zenoh_plugin_ros2dds::route_service_srv] Route Service Server (ROS:/navigate_to_pose/_action/send_goal <-> Zenoh:tb2-1/navigate_to_pose/_action/send_goal): received response for another client: c46c9d06bfe99af1 (me: f9643760349f4104)
[2023-12-20T14:20:46Z WARN  zenoh_plugin_ros2dds::route_service_cli] Route Service Client (ROS:/compute_path_to_pose/_action/get_result <-> Zenoh:tb2-1/compute_path_to_pose/_action/get_result): received error as reply for [9c, 03, 8b, b2, f2, 7e, e5, 28, 27, 00, 00, 00, 00, 00, 00, 00]: Timeout 
```
What I also noticed, is that the Zenoh bridge should be started after the turtlebot bringup, otherwise the nodes will node start up correctly. But I don't know if these two issues are part of the same problem.

Did you find any solution for this?

JEnoch, 2023-12-21:
@samiamlabs : I managed to install Navigation2 as a Dev Container and to run it inside, along `zenoh-bridge-ros2dds`.
However it's using `rolling` and doesn't seems to work properly with my turtlebot3 simulation running on my host with `humble`.  I get some `Failed to parse type hash for topic` errors from `rmw_cyclonedds_cpp` that make me guess some types changed fince `humble`.
If you can provide me a Dev Container for some robot simulation, as well as instructions to reproduce the crash, I would be able to reproduce and debug.

I'm not sure how lifecycle nodes are implemented in ROS 2, and especially if they rely on Services or Actions.
But the "Timeout" log on call to "/compute_path_to_pose/_action/get_result" in @miltzhaw 's comment makes me think all those issues could be related to the bridge being configured with a default timeout of 5 seconds to all queries (i.e. for any call to Service or Action).

Could you please all try to increase this timeout to 10 minutes (or more) on all your bridges:
- either adding `--queries-timeout-default 600` to command line
- either setting [this `queries_timeout/default`](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/2c52d0b32954e66c7759e994511695206b21de65/DEFAULT_CONFIG.json5#L109) parameter in your config file

Note that this setting was added recently, in commit 6e04f60a.

miltzhaw, 2023-12-21:
@JEnoch I just tested what you suggested, changing the timeout value to different values even much higher than 600. While the timeout warning seems to be gone when I use very high values it still doesn't work and I get the following warning on the control bridge before the node dies on the turtlebot:

```
[2023-12-21T13:53:39Z WARN  zenoh_plugin_ros2dds::route_service_srv] Route Service Server (ROS:/navigate_to_pose/_action/send_goal <-> Zenoh:bot1/navigate_to_pose/_action/send_goal): received response for another client: 7823923a07042a85 (me: 5a16ba2a5fcca52e)
```

JEnoch, 2023-12-21:
Thanks for testing.
Could you please send the full logs of the 2 bridges with environment variable: `RUST_LOG=zenoh_plugin=trace`

miltzhaw, 2023-12-22:
I just sent you the full logs per email as they are quite long, thanks

JEnoch, 2023-12-28:
Thanks @miltzhaw! I reviewed your logs.

The `received response for another client` WARN happen when the bridge receives a Service Response message for a a Request initiated by another Service Client than itself. This actually can happen at any time and is not an error, since with Services over DDS the same `Reply` topic is used for replies to **all**  Clients.
This log shouldn't be at WARN level, but rather at DEBUG or TRACE.

On another hand, I saw that each message on `/tf` topic looks to be routed from the Robot to Zenoh, and then back from Zenoh to the Robot! That shouldn't happen, and lead for a Subscriber on `/tf` topic in the Robot to receive each message twice...
Not sure if that's the cause of troubles with nav2. Anyway, that must be fixed and that's what I did in #41.

I'll now do some tests with Lifecycle Nodes to check if they are correctly supported.

JEnoch, 2023-12-29:
There was indeed a bug with Services leading to issue with Lifecycle Nodes: see #43 
That's now fixed in main branch.

@samiamlabs , @btertoolen: could you please test with this fix and tell me if this solved your issue with nav2 ?

samiamlabs, 2024-01-02:
Hi @JEnoch!

I did a quick test with Release #88 of the bridge, and so far nav2 seems to be working!
Will do some more testing this week and get back to you.

Thanks a bunch for fixing this :)

btertoolen, 2024-01-03:
Hi @JEnoch, using the latest build indeed seems to have resolved the issue, so thanks! We will do some more testing, so if we notice anything else I will let you know.

JEnoch, 2024-01-08:
Thanks for your confirmations @btertoolen and @samiamlabs !
I then close this issue. Feel free to open a new one if you get new problems.

I'll do a patch release (`0.10.1-rc2`) with those fixes (#41, #43) in the coming days.

miltzhaw, 2024-01-10:
Sorry for the late reply @JEnoch but I couldn't test it until today. 
The good news is that it worked for me as well now for the tests I have done so far. Thanks a lot for your support, will do more testing in the coming days.
As a note, to make it work I also had to configure the bridges with timestamping enabled as the Warning messages were telling me:

```
  timestamping: {
    enabled: true,
    drop_future_timestamp: false,
  },
```

ciandonovan, 2024-01-15:
Unfortunately I'm still able to reproduce on the latest main (2d9f8c999f497c0d713fef5235c9822970747c13) and using the latest commit on the Humble branch of Nav2 (d29de3dbfcd2513d962357b75aa4db9296bd9620).

Including @miltzhaw's config tweak doesn't help either.

Does the merged fix only work with rolling?

JEnoch, 2024-01-16:
Hi @ciandonovan, the bridge should work with any ROS 2 version (at least from Foxy).

Could you please create a new issue describing what you're experiencing, adding logs (debug level) of bridges, and ideally with a simple way to reproduce.

ciandonovan, 2024-01-16:
@JEnoch this is specifically regarding the Nav2 bringup issue, as described by OP @samiamlabs. Was able to reproduce the original issue with an almost identical setup, will send some reproduceable containers later.

miltzhaw, 2024-01-17:
I actually observed something similar as @ciandonovan is mentioning. 
Namely, I had to start the zenoh bridge after the turtlebot bringup. If the zenoh bridge was already running, the turtlebot bringup would be hanging and not bring all nodes up.

JEnoch, 2024-01-18:
@ciandonovan Thanks for the podman containers. I managed to install and run those.

I see the same log as you. But I don't know what is stuck in nav2 launch.
On bridge side, I see 2 nodes have been discovered (`launch_ros1` and `nav2_container`), but with not so much topics/services except the usual `rosout`, the parameters topics and some node loading/unloading/list services.
What is the launch suppose to do after starting those nodes ?

ciandonovan, 2024-01-18:
This is what it's supposed to look like, and what it looks like when the Zenoh bridge is not running.

```
$ podman run -it --rm --net=host --env RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ghcr.io/ciandonovan/ros2_navigation:main ros2 launch nav2_bringup bringup_launch.py map:=src/navigation2/nav2_bringup/maps/turtlebot3_world.yaml
[INFO] [launch]: All log files can be found below /root/.ros/log/2024-01-18-13-52-28-065297-XXXXXXXXX
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_isolated-1]: process started with pid [57]
[component_container_isolated-1] [INFO] [1705585949.731211807] [nav2_container]: Load Library: /opt/ros/overlay_ws/install/nav2_map_server/lib/libmap_server_core.so
[component_container_isolated-1] [INFO] [1705585949.836903250] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::CostmapFilterInfoServer>
[component_container_isolated-1] [INFO] [1705585949.836977441] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapSaver>
[component_container_isolated-1] [INFO] [1705585949.836991465] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-1] [INFO] [1705585949.837003750] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_map_server::MapServer>
[component_container_isolated-1] [INFO] [1705585949.844047585] [map_server]: 
[component_container_isolated-1] 	map_server lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1705585949.844714698] [map_server]: Creating
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/map_server' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1705585949.846149714] [nav2_container]: Load Library: /opt/ros/overlay_ws/install/nav2_controller/lib/libcontroller_server_core.so
[component_container_isolated-1] [INFO] [1705585949.872954283] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_controller::ControllerServer>
[component_container_isolated-1] [INFO] [1705585949.873055008] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_controller::ControllerServer>
[component_container_isolated-1] [INFO] [1705585949.878496003] [controller_server]: 
[component_container_isolated-1] 	controller_server lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1705585949.906274746] [controller_server]: Creating controller server
[component_container_isolated-1] [INFO] [1705585949.912333473] [local_costmap.local_costmap]: 
[component_container_isolated-1] 	local_costmap lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1705585949.916742305] [local_costmap.local_costmap]: Creating Costmap
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/controller_server' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1705585949.919630998] [nav2_container]: Load Library: /opt/ros/overlay_ws/install/nav2_amcl/lib/libamcl_core.so
[component_container_isolated-1] [INFO] [1705585949.934024675] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated-1] [INFO] [1705585949.934133923] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_amcl::AmclNode>
[component_container_isolated-1] [INFO] [1705585949.939314165] [amcl]: 
[component_container_isolated-1] 	amcl lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1705585949.943059011] [amcl]: Creating
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/amcl' in container '/nav2_container'
[component_container_isolated-1] [INFO] [1705585949.945399930] [nav2_container]: Load Library: /opt/ros/overlay_ws/install/nav2_smoother/lib/libsmoother_server_core.so
[component_container_isolated-1] [INFO] [1705585949.955286781] [nav2_container]: Found class: rclcpp_components::NodeFactoryTemplate<nav2_smoother::SmootherServer>
[component_container_isolated-1] [INFO] [1705585949.955350711] [nav2_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<nav2_smoother::SmootherServer>
[component_container_isolated-1] [INFO] [1705585949.960006677] [smoother_server]: 
[component_container_isolated-1] 	smoother_server lifecycle node launched. 
[component_container_isolated-1] 	Waiting on external lifecycle transitions to activate
[component_container_isolated-1] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[component_container_isolated-1] [INFO] [1705585949.966104652] [smoother_server]: Creating smoother server
[... cut ...]
[component_container_isolated-1] [INFO] [1705585952.544958897] [smoother_server]: Destroying bond (smoother_server) to lifecycle manager.
[component_container_isolated-1] [INFO] [1705585952.545035825] [local_costmap.local_costmap]: Running Nav2 LifecycleNode rcl preshutdown (local_costmap)
[component_container_isolated-1] [INFO] [1705585952.545080102] [local_costmap.local_costmap]: Destroying bond (local_costmap) to lifecycle manager.
[component_container_isolated-1] [INFO] [1705585952.545116358] [lifecycle_manager_navigation]: Running Nav2 LifecycleManager rcl preshutdown (lifecycle_manager_navigation)
```

ciandonovan, 2024-01-22:
I just tested the `fix_62` branch and indeed the previous issue no longer seems to occur.

I can now have the bridge running while launching lifecycle nodes from Nav2 and everything seems to work, nodes come online and the robot navigates correctly.

I will let you know if I come across any other issues, but it does look resolved to me, appreciate your work thank you.

JEnoch, 2024-01-23:
Thanks for your confirmation. The fix is now merged into `main` branch.
All the fixes discussed in this issue will be included in a `0.10.1-rc2` patch version I'll release very soon.

closed 2024-01-23 by JEnoch