eclipse-zenoh/zenoh-plugin-ros2dds #33: [Bug] Gossip discovery fails

Fellfalla, 2023-12-16:
### Describe the bug

I'm trying to setup the a peer-to-peer topology with gossip discovery via a central router. Gossip discovery between 2 robots via a predefined router: robot1 ->  router <- robot2. All of them are running zenoh_bridge_ros2dds locally in the respective mode.

I found 2 issues.

1. Discovery for "scan" topics works only if I set multicast discovery to true. Gossip discovery fails for some reason.
2. When multicast discovery is enable, I can find the respective topic. However, the command `ros2 topic echo` works only the first time. When stopping the echo command and rerunning it, no data arrive anymore. After restarting the local zenoh bridge the echo command works again exactly 1 time.

[zenoh_config](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/files/13693316/zenoh_config)

However, for some reason multicast discovery works, while gossip discovery does not. 

### To reproduce

1. `ros2 run zenoh_bridge_ros2dds zenoh_bridge_ros2dds -c zenoh_config.json5 -e tcp/192.168.0.153:7447`
2. `ros2 topic echo <yourscan>`
3. Router has the same settings, except: mode=router

### System info

- Ubuntu 22.04
- zenoh peer runs directly on the OS
- zenoh router runs withing a devcontainer with --net=host setting.

JEnoch, 2023-12-19:
Hi @Fellfalla ,

I get a 404 error on your `zenoh_config` link: https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/files/13693316/zenoh_config 
Could you please share your config again ?

I understand you're using Docker containers with `--net=host`.
Could you please detail the setup of your containers:
- are bridges running in same or distinct containers than the ROS Nodes ?
- are the containers for the 2 robots running on the same physical host ?
- are the ROS Nodes using distinct ROS_DOMAIN_ID for each robot ?
- is `ROS_LOCALHOST_ONLY=1` set or not ?

Fellfalla, 2023-12-20:
Hey @JEnoch .

I just reuploaded the config file :).
1. Only the bridge with the router config runs in the devcontainer. No nodes are running there. All the nodes are running on the robot. Also the peer mode bridge runs on the robot (nodes and bridge on the robot run without devcontainer). On my service PC, I run a peer in a devcontainer. (no nodes running in the same container, only the `ros2 topic list` command.
2. we have 4 physical systems: 2 robots, 1 router, 1 service laptop. only the 2 latter ones run devcontainers.
3. ROS_DOMAIN_ID=0 for all systems.
4. ros is using localhost, yes. We load follow cyclone configuration 
```
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config
https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <Interfaces>
                <NetworkInterface name="lo" priority="default" multicast="true" />
            </Interfaces>
            <AllowMulticast>default</AllowMulticast>
            <EnableMulticastLoopback>false</EnableMulticastLoopback>
            <MaxMessageSize>65500B</MaxMessageSize>
            <FragmentSize>4000B</FragmentSize>
        </General>
        <Internal>
            <Watermarks>
                <WhcHigh>500kB</WhcHigh>
            </Watermarks>
            <MultipleReceiveThreads>true</MultipleReceiveThreads>    <!-- And this line -->
            <RediscoveryBlacklistDuration>1s</RediscoveryBlacklistDuration>
        </Internal>
    </Domain>
</CycloneDDS>

```

and we load it via a zsh script containing `export CYCLONEDDS_URI=file://$SCRIPT_DIR/cyclone_dds.xml`

imstevenpmwork, 2024-04-10:
Hello @Fellfalla, do you have any updates related to this issue? Let us know (:

Fellfalla, 2024-04-11:
> Hello @Fellfalla, do you have any updates related to this issue? Let us know (:

Making sure the **same zenoh version** is running everywhere, gossip discovery should  work with our described setup.

More tests from our side will follow. I will reopen in case we have new findings.

closed 2024-04-11 by Fellfalla