eclipse-zenoh/zenoh-plugin-ros2dds #42: Improve documentation on integration

fsteff, 2023-12-29:
### Describe the feature

I want to build a zenoh-native application that uses this plugin for compatibility with ROS2, but I could not find any documentation on how exactly the ROS2 services and actions are mapped to Zenoh Queryables.

More specifically, I plan to build a python library for using ROS actions from "outside" of a ROS environment. AFAIK this plugin does exactly what I need for that purpose, but unfortunately am having difficulties finding the required details details in the code.
(as mentioned in "_Easier integration with Zenoh native applications (services and actions are mapped to Zenoh Queryables)_"
 
e.g. are parameters (such as goal_id) of the actions mapped to the Zenoh Selector, or are they just appended to the query as a value?

JEnoch, 2024-01-16:
Hi @fsteff,
Thanks for your suggestion. I didn't had time to address this so far, but I will in the coming weeks.

My plan is to add in this repo some examples of Zenoh applications that are compatible with some of the [ROS 2 demos](https://github.com/ros2/demos) (e.g. `talker`, `listener`, `add_two_ints_client/server`, `fibonacci_action_client/server`...).
I'll do those in Rust and Python as a first step. The other languages will come later (or sooner if anybody would like to contribute 😁)

fsteff, 2024-01-18:
Thanks, that would be very useful.

Maybe I can support the community by releasing the library I'm working on as open source (don't have greenlight from my employer for this yet, but we talked about doing so once it's stable). 
Right now it only supports Zenoh-to-Zenoh actions (mostly a 1:1 implementation of ROS2 actions, using asyncio), but in the future we want to be able to support Zenoh-to-ROS and maybe vice-versa.

JEnoch, 2024-01-24:
That would be great!
I understand this library is in Python using `asyncio`, right ? Do you have plans for other languages ?

Are you aware of the incoming [rmw_zenoh](https://github.com/ros2/rmw_zenoh) ? Once done, it will be useable via `rclpy`.
I guess your library would be lighter than the full `rclpy`+`rcl`+`rmw`+`rmw_zenoh` stack. Do you plan to implement the same full API than `rclpy` or a lighter one ?

fsteff, 2024-01-24:
For now we don't plan to implement it in other languages, though maybe we'll  need to support C++ as well (for future projects).

`rmw_zenoh` was the initial reason why we got interested in zenoh ;-) 

The whole point of building our own library is that it will be lightweight and easy to use. For now we'll only support ROS actions (and maybe services/topics). In the zenoh-to-zenoh usecase (without ROS) the encoding is JSON, which also reduces complexity.