eclipse-zenoh/zenoh-plugin-ros2dds #54: [Bug] Subscribers don't work when the bridge is configured with namespace

guiboro83, 2024-01-12:
### Describe the bug

If a bridge is created with a namespace, ROS subscriber cannot received messages coming from another bridge (using namespace for the publisher side bridge works).

Removing the namespace of the bridge with subscriber resolve the problem.

The problem comes from the namespace which is automatically added to subscribers and that changes the links between subscribers and publishers :

ROS talker ------------- Zenoh network --------- ROS listener (ns = /client)

pub /chatter ----------> chatter -------------------> /chatter
/client/chatter <------- client/chatter <---------- sub /chatter

On this schema, we see that the chatter publisher publish on "chatter" but the chatter listener looking for message on "client/chatter".

The problem is the same for service/client : if there is a namespace on the client side bridge, client and service will never communicate.

### To reproduce

Terminal 1 : ```zenoh-bridge-ros2dds -l tcp/0.0.0.0:7447 -d 10```

Terminal 2 : ```zenoh-bridge-ros2dds -e tcp/127.0.0.1:7447 -n /client -d 11```

Terminal 3 : ```ROS_DOMAIN_ID=10 ros2 run demo_nodes_cpp talker```

Terminal 4 : ```ROS_DOMAIN_ID=11 ros2 run demo_nodes_cpp listener```

The listener never receives any message from the talker but the topic /chatter is visible with a ```ROS_DOMAIN_ID=11 ros2 topic list``` and a ```ROS_DOMAIN_ID=11 ros2 topic echo /chatter``` doesn't get any message.

### System info

- Ubuntu 22.04
- Docker osrf/ros:iron-desktop
- Zenoh bridge 0.10.1-rc

JEnoch, 2024-01-16:
The purpose of a ROS namespace is precisely to add the namespace as a prefix to each topic/service/action name.
For instance, if your run:
 - `ros2 run demo_nodes_cpp talker --ros-args -r __ns:=/bot1`
 - `ros2 topic list`
 You'll see that the topic name is changed to `/bot1/chatter`
 
However, that's a burden for the ROS 2 user to configure all the nodes in a robot with the same namespace `/bot1` to avoid conflicts with the same nodes in another robot `/bot2`.
Therefore, when using `zenoh-bridge-ros2dds`, the user can simplify it's configuration configuring the namespace for the bridge only. All the nodes will communicate inside the robot via DDS without namespace, but outside the robot, the bridge will prefix all topics/services/actions with the namespace over Zenoh.

And for a remote host using another bridge, you have 2 options:
1. use the same namespace configuration for the bridge and no namespace for your nodes.
   I.e. your commands shall be:
   - `zenoh-bridge-ros2dds -l tcp/0.0.0.0:7447 -d 10 -n /client`
   - `zenoh-bridge-ros2dds -l tcp/0.0.0.0:7447 -d 11 -n /client`
   - `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ROS_DOMAIN_ID=10 ros2 run demo_nodes_cpp talker`
   - `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ROS_DOMAIN_ID=11 ros2 run demo_nodes_cpp listener`
2. don't use a namespace for the bridge, but use it for your nodes.
   I.e. your commands shall be:
   - `zenoh-bridge-ros2dds -l tcp/0.0.0.0:7447 -d 10`
   - `zenoh-bridge-ros2dds -l tcp/0.0.0.0:7447 -d 11 -n /client`
   - `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ROS_DOMAIN_ID=10 ros2 run demo_nodes_cpp talker --ros-args -r __ns:=/client`
   - `RMW_IMPLEMENTATION=rmw_cyclonedds_cpp ROS_DOMAIN_ID=11 ros2 run demo_nodes_cpp listener`
   
 You can see this as:
1. you place your 2 bridges in the same namespace, thus the nodes don't have to specify the namespace.
2. you place a bridge in a namespace and the other bridge in the root namespace, thus your nodes served by this later bridge have to explicitly set the namespace they want to communicate with.

guiboro83, 2024-01-23:
Thank you for your answer.
I just wanted to be sure it wasn't a mistake and subscribers work as expected with a namespace in the bridge configuration.

I close the issue.

closed 2024-01-23 by guiboro83