eclipse-zenoh/zenoh-plugin-ros2dds #67: [Bug] Three agents duplicate incoming messages

CatInTheRain, 2024-01-25:
### Describe the bug

Thanks for your amazing work. I was trying the `zenoh_bridge_ros2dds`, with three remote PCs connected on the same Wi-FI. In this demo the `robot 0` publish messages in a topic, and this message is listened by the other two robots (1`` and `2`). The number refers to `ROS_DOMAIN_ID` . Communication sometimes works and sometimes exhibit bugs on the number of messages read by one agent, always the last agent, who duplicates the messages. Is there a way to avoid this, maybe using namespace? Ideally I would like that every agents can publish/subscribe from the same topic without this bug. If this is not a bug could you provide me an example with more than 2 remote pc?

This is the final output 
### Robot 0:
```console
[INFO] [1706202282.021806288] [talker]: Publishing: 'Hello World: 730'
[INFO] [1706202283.021789337] [talker]: Publishing: 'Hello World: 731'
[INFO] [1706202284.021752267] [talker]: Publishing: 'Hello World: 732'
[INFO] [1706202285.021799607] [talker]: Publishing: 'Hello World: 733'
[INFO] [1706202286.021835910] [talker]: Publishing: 'Hello World: 734'
[INFO] [1706202287.021864647] [talker]: Publishing: 'Hello World: 735'
```
### Robot 1:
```console
[INFO] [1706202281.939232374] [listener]: I heard: [Hello World: 730]
[INFO] [1706202281.939652915] [listener]: I heard: [Hello World: 730]
[INFO] [1706202282.939507519] [listener]: I heard: [Hello World: 731]
[INFO] [1706202282.939790751] [listener]: I heard: [Hello World: 731]
[INFO] [1706202283.938842816] [listener]: I heard: [Hello World: 732]
[INFO] [1706202283.942020988] [listener]: I heard: [Hello World: 732]
[INFO] [1706202284.939364677] [listener]: I heard: [Hello World: 733]
[INFO] [1706202284.941937116] [listener]: I heard: [Hello World: 733]
[INFO] [1706202285.939902659] [listener]: I heard: [Hello World: 734]
[INFO] [1706202285.940282093] [listener]: I heard: [Hello World: 734]
[INFO] [1706202286.938968271] [listener]: I heard: [Hello World: 735]
[INFO] [1706202286.939437045] [listener]: I heard: [Hello World: 735]
```
### Robot 2:
```console
[INFO] [1706202282.020367529] [listener]: I heard: [Hello World: 730]
[INFO] [1706202283.019923871] [listener]: I heard: [Hello World: 731]
[INFO] [1706202284.020355711] [listener]: I heard: [Hello World: 732]
[INFO] [1706202285.020997782] [listener]: I heard: [Hello World: 733]
[INFO] [1706202286.021133424] [listener]: I heard: [Hello World: 734]
[INFO] [1706202287.020533780] [listener]: I heard: [Hello World: 735]
```

### To reproduce

## Robot 0 (IP 192.168.0.10)
### Terminal 1
```bash
ROS_DOMAIN_ID=0 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp CYCLONEDDS_URI=/cyclonedds_0.xml
zenoh_bridge_ros2dds -l udp/0.0.0.0:7447 -e udp/192.168.0.22:7447 -e udp/192.168.0.23:7447 --no-multicast-scouting
```
### Terminal 2
```bash
ROS_DOMAIN_ID=0 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp CYCLONEDDS_URI=/cyclonedds_0.xml
ros2 run demo_nodes_cpp talker
```
```console
[INFO] [1706202282.021806288] [talker]: Publishing: 'Hello World: 730'
```

## Robot 1 (IP 192.168.0.22)
### Terminal 1
```bash
ROS_DOMAIN_ID=1 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp CYCLONEDDS_URI=/cyclonedds_1.xml
zenoh_bridge_ros2dds -l udp/0.0.0.0:7447  --no-multicast-scouting
```
### Terminal 2
```bash
ROS_DOMAIN_ID=1 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp CYCLONEDDS_URI=/cyclonedds_1.xml
ros2 run demo_nodes_cpp listener
```
```console
[INFO] [1706202281.939232374] [listener]: I heard: [Hello World: 730]
[INFO] [1706202281.939652915] [listener]: I heard: [Hello World: 730]
```

## Robot 2 (IP 192.168.0.23)
### Terminal 1
```bash
ROS_DOMAIN_ID=2 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp CYCLONEDDS_URI=/cyclonedds_2.xml
zenoh_bridge_ros2dds -l udp/0.0.0.0:7447  --no-multicast-scouting
```
### Terminal 2
```bash
ROS_DOMAIN_ID=2 RMW_IMPLEMENTATION=rmw_cyclonedds_cpp CYCLONEDDS_URI=/cyclonedds_2.xml
ros2 run demo_nodes_cpp listener
```
```console
[INFO] [1706202282.020367529] [listener]: I heard: [Hello World: 730]
```

Robot 1 is the last robot that listen the topic.
the `.xml` configuration are all the same with the only difference in the field `NetworkInterface`

```xml
<CycloneDDS>

 <Domain id="any">

     <General>
         <Interfaces>
             <NetworkInterface name="wlp****"/>
             <!-- For less traffic, force multicast usage on loopback even if not configured.         -->
             <!-- All ROS Nodes and bridges must have this same config, otherwise they won't discover -->
             <!-- <NetworkInterface address="127.0.0.1" multicast="true"/>  -->
         </Interfaces>
         <DontRoute>true</DontRoute>
     </General>
 </Domain>

</CycloneDDS>
```

### System info

- Platform Docker, under image `osrf/ros:humble-desktop`
- Architecture `x86_64`
- Zenoh Version: zenoh-bridge-ros2dds v0.10.1-rc built with rustc 1.70.0 (90c541806 2023-05-31) (built from a source tarball)
zenoh bridge for DDS v0.10.1-rc
- Zenoh was build as a ROS2 package

CatInTheRain, 2024-01-26:
> The fix for it has been pushed in recent patch release [0.10.1-rc.2](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/releases/tag/0.10.1-rc.2). Could you please try with it and let me know the result ?

You are right it finally works! Thank you, and sorry for not see the new patch release. I am working on multi robot application under Wi-Fi and under transfer of big data size. It seems to work really nice for my experiments, respect to classical DDS!  I will cite your amazing repositories in my works!

One last question if I don't bother you: I am working on a program that can monitor the network traffic (like [fast-dds-monitor](https://www.eprosima.com/index.php/products-all/eprosima-fast-dds-monitor)) to evaluate the improvements respect to other middleware for ROS2. 
Are there any possible solutions under working in progress? Otherwise, if I can develop something initial and experimental to monitor the traffic of some ROS 2 topics I can update you. Thanks again for the patience!

JEnoch, 2024-01-26:
Thanks for your confirmation! I'm closing this ticket then.

> One last question if I don't bother you: I am working on a program that can monitor the network traffic (like [fast-dds-monitor](https://www.eprosima.com/index.php/products-all/eprosima-fast-dds-monitor)) to evaluate the improvements respect to other middleware for ROS2.
Are there any possible solutions under working in progress? Otherwise, if I can develop something initial and experimental to monitor the traffic of some ROS 2 topics I can update you

What we plan to add in the bridge is some internal measurement of frequency and throughput per-Topic (measurement point when routing between DDS and Zenoh). Actually, one of our user volunteered to implement this feature. You can see our discussion in Discord here: https://discord.com/channels/914168414178779197/1195411111550386360/1197575114305323018
Maybe you can check with him if you would like to contribute ?

For measurement of global traffic of a Zenoh system, we have some tooling but the measurements are per link between Sessions, not per-topic.

Finally, for a more fine-grained analysis of zenoh protocol and traffic, you could have a look to our [Wireshark dissector](https://github.com/ZettaScaleLabs/zenoh-dissector).

closed 2024-01-26 by JEnoch