eclipse-zenoh/zenoh-plugin-ros2dds #70: Zenoh-only examples communicating with ROS Nodes via the bridge

JEnoch, 2024-01-29:
### Describe the feature

The goal is to provide examples of applications using the Zenoh APIs to communicate with ROS 2 Nodes via `zenoh-bridge-ros2dds`. At minimum, those examples shall include some which are compatible with the [ROS 2 demos](https://github.com/ros2/demos).

JEnoch, 2024-02-06:
Draft PR in progress in [examples](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/tree/examples/examples) branch.

ping @fsteff !

fsteff, 2024-02-08:
thanks @JEnoch !
Great, straightforward implementation. Somehow I expected that `services and actions are mapped to Zenoh Queryables` means more ROS-specific functionality built-in... I guess that simply confused my overcomplicated thinking brain.

JEnoch, 2024-02-08:
Which _built-in ROS-specific functionalities_ did you expect for services and actions ?

uupks, 2024-04-02:
I implemented a  [zenoh-pico-example](https://github.com/uupks/zenoh-plugin-ros2dds/tree/zenoh_pico_examples/examples/zenoh-pico) based on cyclonedds cdrstream, but now there are several problems: 

1. add_two_ints_client

  client send reqeust with a = 123456(0x01_E240), b = 654321(0x09_FBF1)
 
  while server received request with a = 530239482494992(0x01_E240_0000_0010), b = 2810287296086016(0x09_FBF1_0000_0000)

  client received response with sum = 3340526778581008(0x0B_DE31_0000_0010), actual sum = 777777(0x0B_DE31)

  ```
  Sending Query 'add_two_ints'... with a = 123456, b = 654321
  0 0X1 0 0 0X10 0 0 0 0X40 0XE2 0X1 0 0 0 0 0 0XF1 0XFB 0X9 0 0 0 0 0 
  >> Received ('add_two_ints': 3340526778581008)
  ```
  
  ```
  [INFO] [1712021043.259895789] [add_two_ints_server]: Incoming request
  a: 530239482494992 b: 2810287296086016
  ```

2. listener

```
dds_is_get1: Assertion `is->m_index < is->m_size' failed.
```

I don't know if it's a problem with my use of cdrstream or a problem with bridge...

JEnoch, 2024-04-02:
@uupks The issue is in your usage of cdrstream:
- You define a CDR header indicating your payload will be encoded as little-endian (CDR_LE representation, see [DDSI specification §10.5](https://www.omg.org/spec/DDSI-RTPS/2.5/PDF))
- But you use cdrstream's big-endian encoding operation: [`dds_stream_write_sample()`](https://github.com/uupks/zenoh-plugin-ros2dds/blob/f95351ea6a0d711641d40d6916181196d0d67034/examples/zenoh-pico/add_two_ints_client.c#L173C16-L173C39). 

You should call `dds_stream_write_sampleLE()` instead.

Similarly, [when deserializing the reply](https://github.com/uupks/zenoh-plugin-ros2dds/blob/zenoh_pico_examples/examples/zenoh-pico/add_two_ints_client.c#L34-L50), you should check the 4 bytes CDR header to check if the encoding is little or big endian, and then call `dds_stream_read_sampleLE()` or `dds_stream_read_sample()` accordingly.

uupks, 2024-04-02:
I tested both `dds_stream_write_sampleLE` and `dds_stream_write_sampleBE`, but not working.

- dds_stream_write_sampleLE
```
Sending Query 'add_two_ints'... with a = 123456, b = 654321
Send : 0 0X1 0 0 0X10 0 0 0 0X40 0XE2 0X1 0 0 0 0 0 0XF1 0XFB 0X9 0 0 0 0 0 
Received : 0 0X1 0 0 0X10 0 0 0 0X31 0XDE 0XB 0 
>> Received ('add_two_ints': 3340526778581008)
>> Received query final notification
```

- dds_stream_write_sampleBE
```
Sending Query 'add_two_ints'... with a = 123456, b = 654321
Send : 0 0X1 0 0 0 0 0 0X10 0 0 0 0 0 0X1 0XE2 0X40 0 0 0 0 0 0X9 0XFB 0XF1
Received : 0 0X1 0 0 0 0X1 0XE2 0X50 0 0 0 0 
>> Received ('add_two_ints': 1356988672)
>> Received query final notification
```

and there is no dds_stream_read_sampleLE() with the latest [cdrstream](https://github.com/eclipse-cyclonedds/cyclonedds/blob/master/src/core/cdr/src/dds_cdrstream.c), only the `dds_stream_read_sample()`

PR #104 (2024-04-22): Added ROS2-like Rust Examples
This adds ROS2-like examples written in Rust for:
* Topics listener
 * Topics talker
 * Services client
 * Actions client

Plus slight modifications to the Python ones. Mostly related to directory structure and logs.

Resolves #70 
Main PR: https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/80