eclipse-zenoh/zenoh-plugin-ros2dds #78: [Bug] Upgrading debian/ubuntu packaged removes the zenoh-bridge-ros2dds user and /etc/zenoh-bridge-ros2dds directory.

aosmw, 2024-02-02:
### Describe the bug

Using apt to install an update of zenoh-bridge-ros2dds triggers a removal of the zenoh-bridge-ros2dds user and the /etc/zenoh-bridge-ros2dds directory.

The upgrade fails if the service is in use by the zenoh-bridge-ros2dds user.   

The upgrade should not remove the existing conf.json5 file.  (It contained hard earned configuration settings :/ ).

The default file config file could possibly installed in /lib/share/zenoh-bridge-ros2dds/default_conf.json5 

If a /etc/zenoh-bridge-ros2dds/conf.json5 does not exist then it might fallback to the installed /lib/share/zenoh-bridge-ros2dds/default_conf.json5.  I am making this up. There is probably a proper debian way to handle configuration files and service restarts during upgrades.

### To reproduce

1. Attempt to upgrade a running zenoh-bridge-ros2dds service with a customised config.
2. The upgrade will fail until the service is stopped.
3. Any previous config will be replaced with a default config.

### System info

- Ubuntu 22.04.3 LTS
- [2024-02-02T03:05:46Z INFO  zenoh_bridge_ros2dds] zenoh-bridge-ros2dds v0.10.1-rc.2 built with rustc 1.72.0 (5680fa18f 2023-08-23

JEnoch, 2024-02-02:
Sorry about your lost config!
We need to have a look on how a Debian package is supposed to handle updates in a proper way.

JEnoch, 2024-04-23:
The proper way seems to declare the configuration file in a [`conffiles`](https://www.debian.org/doc/manuals/maint-guide/dother.en.html#conffiles) file within the Debian package.
Addressed in #121.

PR #121 (2024-04-23): Debian pkg: add a conffiles to avoid overrwriting a modified conf.json5
Fix #78 

This PR adds a [`conffiles`](https://www.debian.org/doc/manuals/maint-guide/dother.en.html#conffiles) file in the Debian package for `zenoh-bridge-ros2dds` that references the `/etc/zenoh-bridge-ros2dds/conf.json5` configuration file.
Thus, if modified, it will not been overwritten when upgrading the package.
closed 2024-04-23 by Mallets