eclipse-zenoh/zenoh-plugin-ros2dds #84: [Bug] Unable to open listener tcp/[::]:0

Daylien, 2024-02-26:
### Describe the bug

Hey!

I'm trying to run `zenoh-bidge-ros2dds` binary, but the binary won't start because the TCP listener can't be bound. The strange thing is, that the listener is trying to bind it to the `[::]:0` local IPv6 address, even when specifying an IPv4 address.
I'm running an unmodified Ubuntu 23.10, but now the strange thing. When using Ubuntu 22.04 or EndeavourOS, Zenoh is starting successfully. So I suspect a network interface misbehaviour (maybe due to misconfiguration or an Ubuntu specific issue with resolving the network interface).

Nevertheless, maybe someone is able to reproduce this issue. 

### To reproduce

1. Get Ubuntu 23.10
2. Get zenoh-bridge-ros2dds
3. Start the zenoh-bridge-ros2dds bridge with or without parameters
4. Crash with the following error message:

```
ERROR zenoh::net::runtime::orchestrator] Unable to open listener tcp/[::]:0: 
Can not create a new TCP listener bound to tcp/[::]:0: [[::]:0: Address family not supported by protocol (os error 97) 
at /home/user/.cargo/git/checkouts/zenoh-cc237f2570fab813/42f9384/io/zenoh-links/zenoh-link-tcp/src/unicast.rs:222.] 
at /home/user/.cargo/git/checkouts/zenoh-cc237f2570fab813/42f9384/io/zenoh-links/zenoh-link-tcp/src/unicast.rs:305.

thread 'main' panicked at 'called `Result::unwrap()` on an `Err` value: 
Can not create a new TCP listener bound to tcp/[::]:0: [[::]:0: Address family not supported by protocol (os error 97) 
at /home/user/.cargo/git/checkouts/zenoh-cc237f2570fab813/42f9384/io/zenoh-links/zenoh-link-tcp/src/unicast.rs:222.]
at /home/user/.cargo/git/checkouts/zenoh-cc237f2570fab813/42f9384/io/zenoh-links/zenoh-link-tcp/src/unicast.rs:305.', zenoh-bridge-ros2dds/src/main.rs:77:62```

### System info

- Platform: Ubuntu 23.10
- CPU: Intel i7-1255U (x86_64)

JEnoch, 2024-02-27:
It was a deliberate choice in Zenoh to bind by default on TCP/IPv6 ANY address with a port chosen by the OS, i.e. `[::]:0` (see https://github.com/eclipse-zenoh/zenoh/pull/366).
It looks like your Ubuntu 23.10 is configured without the support of IPv6.

Still, you should be able to force the binding to TCP/IPv4 ANY address as such:
`zenoh-bidge-ros2dds  -l tcp/0.0.0.0:0`

Daylien, 2024-02-27:
Hi, thanks for your answer!

Yeah, the `zenoh-bridge-ros2dds -l tcp/127.0.0.1:7447` is working fine, but connecting to a binded port using the following:
`./zenoh-bridge-ros2dds -e tcp/127.0.0.1:7447` crashes the zenoh bridge. Or am I missing something for the connection part?

JEnoch, 2024-03-01:
`-e tcp/127.0.0.1:7447` specifies the bridge it should try to **connect to** 127.0.0.1 (localhost IP)  on port 7447.
If you don't have another Zenoh router or peer running on the same host and which is **listening on** this port (via `-l 127.0.0.1:7447` or `-l tcp/0.0.0.0:7447`) then the bridge can't establish the connection.
If the bridge is in `peer` mode (by default), it doesn't harm - the bridge keeps periodically trying and failing silently.
If the bridge is in `client` mode, it exits with an error after a timeout. This is because a Zenoh client **must** have 1 (and only 1) connection established to a router or a peer.

Daylien, 2024-03-01:
Thank you, I have solved the problem. The reference to client mode makes sense. Sorry for the inconvenience. I am closing the issue.

Thanks again!

closed 2024-03-01 by Daylien