eclipse-zenoh/zenoh-plugin-ros2dds #134: [Bug] Docker nightly outdated

Timple, 2024-05-03:
### Describe the bug

https://hub.docker.com/r/eclipse/zenoh-bridge-ros2dds/tags

The nightly is two months old, and yet there are newer tags pushed. 
Or am I missing a policy here?

### To reproduce

Go to https://hub.docker.com/r/eclipse/zenoh-bridge-ros2dds/tags, check the date of nightly build :slightly_smiling_face: 

### System info

Any platform using docker?

JEnoch, 2024-05-03:
Thanks for reporting this.
It seems to be a regression following a complete change of the CI organisation. This issue will be addressed soon here: https://github.com/eclipse-zenoh/ci/issues/123

diogomatsubara, 2024-05-16:
This has been fixed by the latest rc.3 release. The docker tags are now properly tagged with nightly for the nightly release and the latest for the rc3 release.

closed 2024-05-16 by diogomatsubara