eclipse-zenoh/zenoh-plugin-ros2dds #139: [Bug]  Deb install do not found systemctl even if it's present

ggari-robotnik, 2024-05-06:
### Describe the bug

The installation of the zenoh-bridge-ros2dds 0.11.0-rc2 fails to found systemctl command, even if the command is found

### To reproduce

```
command -v systemctl
/usr/bin/systemctl
echo $?
0
```
```bash
sudo dpkg -i zenoh-plugin-ros2dds_0.11.0-rc.2_amd64.deb
....
Setting up zenoh-bridge-ros2dds (0.11.0-rc.2) ...
WARNING: 'systemctl' not found - cannot install zenoh-bridge-ros2dds as a service.
/usr/bin/systemctl
``

### System info

- Platform: ubuntu 22.04.3 server
- CPU:  Intel(R) Core(TM) i7-10710U
- Zenoh version: 0.11.0-rc2 (deb)

Mallets, 2024-05-06:
Tagging @diogomatsubara

diogomatsubara, 2024-05-07:
Hi @ggari-robotnik, thanks for the bug report!

I was unable to reproduce the exact issue you mentioned as it seems our debian repository is broken at the moment. I got a different error trying to install the `zenoh-bridge-ros2dds` package via apt on a pristine Ubuntu container. I'll debug this further today but I'm curious why you choose to install the `zenoh-bridge-ros2dds` via dpkg instead of apt. Did you get an error installing the `zenoh-*` packages on Ubuntu via apt and then tried to install the individual package manually using dpkg?

ggari-robotnik, 2024-05-08:
Dear @diogomatsubara I use dpkg because sometimes in the past the zenoh apt repo has mixed versions. I wanted make sure that I have that version.

diogomatsubara, 2024-05-16:
Hi @ggari-robotnik , we just released `0.11.0-rc.3` and I tried the debian repository on a pristine ubuntu install and couldn't reproduce the issue. Could you give it a try? Another thing to consider is if there are any security profile applied to your OS which would make systemctl inacessible or not in the $PATH for the install process.

meetgandhi-dev, 2025-03-03:
Hey @diogomatsubara 
I found the similar issue on my Ubuntu 22.04 host.
I believe the following PR fixes the same: https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/416