eclipse-zenoh/zenoh-plugin-ros2dds #153: [Bug]

Fellfalla, 2024-05-25:
### Describe the bug

> [WARN] [1716629393.254861174] [rmw_cyclonedds_cpp]: Failed to parse type hash for topic 'rt/india/move_state_goal_pose' with type 'geometry_msgs::msg::dds_::PoseStamped_' from USER_DATA '(null)'.

The example script to reproduce with according profile of `move_to_goal_pose` is 
```python
import rclpy
from rclpy.qos import QoSProfile, QoSReliabilityPolicy, QoSHistoryPolicy
from geometry_msgs.msg import PoseStamped

rclpy.init()
node = rclpy.create_node('state_control', automatically_declare_parameters_from_overrides=True)

qos_profile = QoSProfile(
    reliability=QoSReliabilityPolicy.RELIABLE,
    history=QoSHistoryPolicy.KEEP_LAST,
    depth=1
)

pub = node.create_publisher(PoseStamped, 'move_state_goal_pose', qos_profile=qos_profile)

rclpy.spin(node)
```

This error happens in conjunction with following `cyclone_dds.xml` configuration 
```xml
<?xml version="1.0" encoding="UTF-8" ?>
<CycloneDDS xmlns="https://cdds.io/config" xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance" xsi:schemaLocation="https://cdds.io/config
https://raw.githubusercontent.com/eclipse-cyclonedds/cyclonedds/master/etc/cyclonedds.xsd">
    <Domain id="any">
        <General>
            <Interfaces>
                <NetworkInterface name="lo" priority="default" multicast="true" />
            </Interfaces>
        </General>
    </Domain>
</CycloneDDS>
```

* Note! Changing the NetworkInterface from `lo` to `wlo1` gets rid of the error message

### To reproduce

1. launch zenoh via `zenoh-bridge-ros2dds`
2. run `CYCLONEDDS_URI="cyclone_dds.xml" python3 tests.py` based on the provided example python code

### System info

- zenoh bridge for DDS v0.10.1-rc built with rustc 1.72.0 (5680fa18f 2023-08-23)
- `Linux kilo 6.5.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
- ROS_DISTRO: iron irwini inside modified `osrf/ros:iron-desktop-full` container image

Rayman, 2024-06-07:
I have the same issue. Any idea what could cause this?

JEnoch, 2024-10-18:
This is the same issue than https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/issues/290.
Closing this one as duplicate.