eclipse-zenoh/zenoh-plugin-ros2dds #190: Latest version broken on Humble [Bug]

martincerven, 2024-07-16:
### Describe the bug

Latest version (1.0.0-alpha.4) of `zenoh-bridge-ros2dds` installed from apt on 22.04 & Humble is broken even with simple:
`ros2 run demo_nodes_cpp talker` & `ros2 run demo_nodes_cpp listener` and prints errors:

```
1721139897.221382 [42] 	recvUC: malformed packet received from vendor 1.16 state parse:INFO_TS <52545053 02010110 26de1001 97a33730 d50177d6 0e010c00 01108b96 de640447 @0xac 09010800 d0819666 f97070a2 1505a800 00001000 00001404 00001303 00000000 07000000 d0819666 22a47425 0a000000 2f6c6973 74656e65 72000000 01000000> (note: maybe partially bswap'd) smid 0x9 flags 0x1 otnh 8
```

I was trying zenoh bridge with cyclone+iceoryx & images & pointclouds and had installed older version of bridge on one machine and latest on the other, it threw just yet another type of error which I didn't wrote down.

It also printed these malformed packet errors which I at first thought is because of pointclouds or iceoryx or whatnot but then I downloaded [older version of bridge](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/releases/tag/0.11.0-rc.3) on both machines and it works even with images, pointclouds and iceoryx.

Here are exaples of configs that I use over wifi:
```
{
    mode: "peer",
    connect: { endpoints: ["tcp/10.42.0.1:7447"] },
    plugins: {
      ros2dds: {
        id: "remote",
        namespace: "/bot1",
        deny: {
          publishers: [".*/_internal/.*"],
          subscribers: [".*/_internal/.*"],
          service_servers: [".*/_internal/.*"],
          service_clients: [".*/_internal/.*"],
          action_servers: [".*/_internal/.*"],
          action_clients: [".*/_internal/.*"]
        },
        pub_max_frequencies: [
          ".*/image_raw=10",
          ".*/camera_info=10",
           ".*/overlay=10",
           ".*/points=10",
        ],
      },
      rest: { http_port: 8000 }
    }
  }
```

### To reproduce

1. `zenoh-bridge-ros2dds` on  machine A
2. `zenoh-bridge-ros2dds` on  machine B
3. `ros2 run demo_nodes_cpp talker` on machine A
4. `ros2 run demo_nodes_cpp listener` on machine B
5. close one or other node or wait and errors will appear

### System info

Ubuntu 22.04
Humble
CycloneDDS
Nvidia Jetson Orin devices
Wifi hotspot

franz6ko, 2024-07-16:
I am having the very same problem with ROS2 Foxy using CycloneDDS.

I have a router configured to run over my ROS_DOMAIN 0 and a client over my ROS_DOMAIN 1

Router:
[image]

Client:
[image]

If I comment the position topic on the client, a ros2 topic list command list the topic but an echo wont publish it.
If I uncomment the position topic on the client, a ros2 topic echo will give the same error messages @martincerven is having:

[image]

aosmw, 2024-07-16:
I can also confirm.

I rolled back to 0.11.0-stable and marked it with a apt-mark hold.

The forced purging of the package is also necessary for me due to bugs in the debian maintainer scripts.

sudo -i
systemctl status zenoh-bridge-ros2dds.service
systemctl stop zenoh-bridge-ros2dds.service
apt-mark unhold zenoh-bridge-ros2dds
dpkg --purge --force-all zenoh-bridge-ros2dds
id -u zenoh-bridge-ros2dds &> /dev/null || useradd -r -s /bin/false zenoh-bridge-ros2dds
apt install zenoh-bridge-ros2dds=0.11.0-stable
apt-mark hold zenoh-bridge-ros2dds
systemctl start zenoh-bridge-ros2dds.service
journalctl -ru zenoh-bridge-ros2dds.service

aosmw, 2024-07-16:
Possibly related - https://github.com/eclipse-zenoh/zenoh-pico/pull/530

sumitpaulde, 2024-07-23:
I am having this error.
```
2024-07-23T13:23:36.387283Z ERROR             net-0 ThreadId(20) zenoh_transport::unicast::establishment::open: Received a close message (reason INVALID) in response to an InitSyn on: tcp/192.168.2.102:42828 => tcp/192.168.2.108:7447 at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-transport-1.0.0-alpha.4/src/unicast/establishment/open.rs:239.
```
When I am trying to subscribe to the ros2 topic with zenoh using zenoh-plugin-ros2dds between two physical nodes.

evshary, 2024-08-12:
Hi @sumitpaulde I didn't manage to reproduce your error. Would you mind creating another issue and providing the reproducible step? Thank you.
@JEnoch I guess the issue can be closed after https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/191

JEnoch, 2024-08-12:
> I am having this error.
>
> ```
> 2024-07-23T13:23:36.387283Z ERROR             net-0 ThreadId(20) zenoh_transport::unicast::establishment::open: Received a close message (reason INVALID) in response to [...]

Hi @sumitpaulde,
this error message indicates that 2 Zenoh processes are failing to establish a connection with each other because they're not using the same protocol version (`INVALID`  message in response to `InitSyn`).
Please, make sure to use the exact same version for all your `zenoh-plugin-ros2dds`.

JEnoch, 2024-08-12:
> Hi @sumitpaulde I didn't manage to reproduce your error. Would you mind creating another issue and providing the reproducible step? Thank you. @JEnoch I guess the issue can be closed after #191

Sure! Don't know why it was not automatically closed following the merge, since the 1st PR's comment is telling Github it fixes this issue...

PR #191 (2024-07-19): Fix regression when porting to Zenoh 1.0.0
Fix #190 

An error slipped in while porting to Zenoh 1.0.0:
The conversion of `DDSRawSample` into the new `ZBytes` was wrong and truncating the 4 bytes CDR Header.
As a consequence a the bridge routing from DDS to Zenoh was sending an invalid CDR buffer to the remote bridge that published it as such over DDS. When a ROS Node received this invalid message CycloneDDS detected an malformed packet.

This PR also reverts some changes from #169 which, while removing some `unwrap()`, added extra copies and therefore had a performance impact.
closed 2024-08-12 by JEnoch