eclipse-zenoh/zenoh-plugin-ros2dds #202: [Bug]  configuration error: unknown field `id`  -  1.0.0~alpha.5-1

martincerven, 2024-08-01:
### Describe the bug

Latest binary  zenoh-bridge-ros2dds produces `unknown field id` error with config that works with `sudo apt install zenoh-bridge-ros2dds=0.11.0-stable` version. 

```
{
    mode: "peer",
    listen: { endpoints: ["tcp/10.42.0.1:7447"] },
    plugins: {
      ros2dds: {
        id: "bot1",
        namespace: "/bot1",
        deny: {
          publishers: [".*/_internal/.*"],
          subscribers: [".*/_internal/.*"],
          service_servers: [".*/_internal/.*"],
          service_clients: [".*/_internal/.*"],
          action_servers: [".*/_internal/.*"],
          action_clients: [".*/_internal/.*"]
        },
        pub_max_frequencies: [
          ".*/image_raw=10",
          ".*/camera_info=10",
          ".*/overlay=10",
          ".*/points=5",

        ],
      },
      rest: { http_port: 8000 }
    }
  }

```

### To reproduce

1. zenoh-bridge-ros2dds -c <path to config>
2. see error:
```
2024-08-01T07:22:21.441317Z  INFO main ThreadId(01) zenoh_bridge_ros2dds: zenoh-bridge-ros2dds v1.0.0-alpha.5
2024-08-01T07:22:21.441838Z  INFO main ThreadId(01) zenoh_bridge_ros2dds: Zenoh Config { id: 5ca1735fcb6ebfeed4ac52bc888bf07f, metadata: Null, mode: Some(Peer), connect: ConnectConfig { timeout_ms: None, endpoints: Unique([]), exit_on_failure: None, retry: None }, listen: ListenConfig { timeout_ms: None, endpoints: Unique([tcp/10.42.0.1:7447]), exit_on_failure: None, retry: None }, scouting: ScoutingConf { timeout: None, delay: None, multicast: ScoutingMulticastConf { enabled: None, address: None, interface: None, ttl: None, autoconnect: None, listen: None }, gossip: GossipConf { enabled: None, multihop: None, autoconnect: None } }, timestamping: TimestampingConf { enabled: Some(Unique(true)), drop_future_timestamp: None }, queries_default_timeout: None, routing: RoutingConf { router: RouterRoutingConf { peers_failover_brokering: None }, peer: PeerRoutingConf { mode: None } }, aggregation: AggregationConf { subscribers: [], publishers: [] }, transport: TransportConf { unicast: TransportUnicastConf { accept_timeout: 10000, accept_pending: 100, max_sessions: 1000, max_links: 1, lowlatency: false, qos: QoSUnicastConf { enabled: true }, compression: CompressionUnicastConf { enabled: false } }, multicast: TransportMulticastConf { join_interval: Some(2500), max_sessions: Some(1000), qos: QoSMulticastConf { enabled: false }, compression: CompressionMulticastConf { enabled: false } }, link: TransportLinkConf { protocols: None, tx: LinkTxConf { sequence_number_resolution: U32, lease: 10000, keep_alive: 4, batch_size: 65535, batching: true, queue: QueueConf { size: QueueSizeConf { control: 1, real_time: 1, interactive_high: 1, interactive_low: 1, data_high: 2, data: 4, data_low: 2, background: 1 }, congestion_control: CongestionControlConf { wait_before_drop: 1000 }, backoff: 100 }, threads: 2 }, rx: LinkRxConf { buffer_size: 65535, max_message_size: 1073741824 }, tls: TLSConf { root_ca_certificate: None, server_private_key: None, server_certificate: None, client_auth: None, client_private_key: None, client_certificate: None, server_name_verification: None, root_ca_certificate_base64: None, server_private_key_base64: None, server_certificate_base64: None, client_private_key_base64: None, client_certificate_base64: None }, unixpipe: UnixPipeConf { file_access_mask: None } }, shared_memory: ShmConf { enabled: false }, auth: AuthConf { usrpwd: UsrPwdConf { user: None, password: None, dictionary_file: None }, pubkey: PubKeyConf { public_key_pem: None, private_key_pem: None, public_key_file: None, private_key_file: None, key_size: None, known_keys_file: None } } }, adminspace: AdminSpaceConf { enabled: true, permissions: PermissionsConf { read: true, write: false } }, downsampling: [], access_control: AclConfig { enabled: false, default_permission: Deny, rules: None, subjects: None, policies: None }, plugins_loading: PluginsLoading { enabled: true, search_dirs: None }, plugins: Object {"rest": Object {"http_port": Number(8000)}, "ros2dds": Object {"deny": Object {"action_clients": Array [String(".*/_internal/.*")], "action_servers": Array [String(".*/_internal/.*")], "publishers": Array [String(".*/_internal/.*")], "service_clients": Array [String(".*/_internal/.*")], "service_servers": Array [String(".*/_internal/.*")], "subscribers": Array [String(".*/_internal/.*")]}, "domain": Number(42), "id": String("bot1"), "namespace": String("/bot1"), "pub_max_frequencies": Array [String(".*/image_raw=10"), String(".*/camera_info=10"), String(".*/overlay=10"), String(".*/points=5")], "ros_localhost_only": Bool(false)}} }
2024-08-01T07:22:21.441967Z  INFO main ThreadId(01) zenoh::net::runtime: Using ZID: 5ca1735fcb6ebfeed4ac52bc888bf07f
2024-08-01T07:22:21.442337Z  INFO main ThreadId(01) zenoh::api::loader: Starting required plugin "rest"
2024-08-01T07:22:21.443422Z  INFO main ThreadId(01) zenoh::api::loader: Successfully started plugin rest from "__static_lib__"
2024-08-01T07:22:21.443433Z  INFO main ThreadId(01) zenoh::api::loader: Finished loading plugins
2024-08-01T07:22:21.443439Z  INFO main ThreadId(01) zenoh::api::loader: Starting required plugin "ros2dds"
thread 'main' panicked at /cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-1.0.0-alpha.5/src/api/loader.rs:96:21:
Plugin "ros2dds" failed to start: Plugin `ros2dds` configuration error: unknown field `id` at zenoh-plugin-ros2dds/src/lib.rs:128.
stack backtrace:
   0: rust_begin_unwind
             at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:645:5
   1: core::panicking::panic_fmt
             at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
   2: zenoh::net::runtime::RuntimeBuilder::build::{{closure}}
   3: zenoh_bridge_ros2dds::main::main::{{closure}}
   4: zenoh_bridge_ros2dds::main
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
Aborted (core dumped)
```

### System info

Ubuntu 22.04
Humble
CycloneDDS
Nvidia Jetson Orin devices
Wifi hotspot

evshary, 2024-08-12:
Hi @martincerven I think this is intended in [this PR](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/179)
You could take a look at [the related issue](https://github.com/eclipse-zenoh/zenoh/issues/1190) for more detail.

martincerven, 2024-08-12:
Yeah, but that's the dev branch, the  [example config](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/main/DEFAULT_CONFIG.json5) in main still mentions id.

evshary, 2024-08-13:
> Yeah, but that's the dev branch, the [example config](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/main/DEFAULT_CONFIG.json5) in main still mentions id.

So you're using 0.11.0? Because I saw the log you provided is using 1.0.0
```
2024-08-01T07:22:21.441317Z  INFO main ThreadId(01) zenoh_bridge_ros2dds: zenoh-bridge-ros2dds v1.0.0-alpha.5
```

martincerven, 2024-08-13:
I'm comparing `0.11.0` which is version that worked for me, with latest binary one that I get with `apt install` , which is presumably build from main branch, and as you correctly saw in the logs, two weeks ago it was  `zenoh-bridge-ros2dds v1.0.0-alpha.5`

So you say that id variable was removed in **dev** branch,
And I'm saying that id is **still there** in **main** branch [in example config](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/862fe5fd9b817a39589a57d272a52bbd4d31fba7/DEFAULT_CONFIG.json5#L18), which is branch that is presumably used for building binaries for debian packages.

I don't know if that PR was merged or not in main, but one of files changed was example config, and main still has the old config with id variable.

@JEnoch Was the id removed from main without updating example config?

JEnoch, 2024-08-13:
No, the `id` was removed only in `dev/1.0.0` branch for the reasons explained in https://github.com/eclipse-zenoh/zenoh/issues/1190. The example config has been updated in this branch.
It has not yet been merged to `main` branch.

However as we do releases of `1.0.0-alpha.X` versions from `dev/1.0.0` branch, the Debian packages are also made available in our repo. Unfortunately Debian considers `1.0.0-alpha.X > 0.11.Y` and thus installs a `1.0.0-alpha.X` by default.
Hence, you get `zenoh-bridge-ros2dds v1.0.0-alpha.5` in your logs.

Sorry for the inconvenience. But you'll have to remove this `id` anyway when upgrading to `1.0.0`.

JEnoch, 2024-08-30:
I'm closing this issue since we won't fix it.
You can define you're own `id` for the bridge as an hexadecimal string in the Zenoh config part. e.g.:
```json5
{
  plugins: {
    ros2dds: {
      // ...
    }
  },
  
  /// The identifier (as unsigned 128bit integer in hexadecimal lowercase - leading zeros are not accepted)
  /// that zenoh runtime will use.
  /// If not set, a random unsigned 128bit integer will be used.
  /// WARNING: this id must be unique in your zenoh network.
  id: "1234567890abcdef",
}
```

#234 is completing this change, remapping the `-i`/`--id` option to the Zenoh ID config.

closed 2024-08-30 by JEnoch