eclipse-zenoh/zenoh-plugin-ros2dds #209: [Bug] zenoh router takes 100% CPU after closing transient local subscriber

JMvanBruggen, 2024-08-06:
### Describe the bug

Zenoh router takes 100% CPU usage after closing a transient local subscriber. 

### To reproduce

1. Start a router: ``zenoh-bridge-ros2dds -m router``
2. Start a transient local subscriber. I recreated the [rclcpp_minimal_subscriber example](https://github.com/ros2/examples/blob/humble/rclcpp/topics/minimal_subscriber/lambda.cpp), only changing the QoS to ``rclcpp::QoS(1).transient_local()``
3. Stop the subscriber

Router keeps functioning but CPU usage spikes to 100% and stays that way. I have to restart the router to fix it. But everytime a node with a transient local subscriber fails or respawns this happens. I first noticed this on the client side by the way, but that has an extra reproduction step, not sure if that is of importance: 

1. Start a router: ``zenoh-bridge-ros2dds -m router``
2. Start a client ``zenoh-bridge-ros2dds -e tcp/ROBOT_IP:7447 -m client``
3. Start a transient local subscriber on the client side.
4. Stop the subscriber

### System info

- Ubuntu 22:04
- ROS Humble
- zenoh-bridge-ros2dds  1.0.0~alpha.5-1 amd64

aosmw, 2024-08-08:
I see 100% cpu often but don't know what triggers it.

I flamegraphed it when I caught it. Maybe we can compare flamegraphs and give the maintainers a hint.

```
# Install perf
sudo apt install linux-tools-generic
cargo install flamegraph

# Read the help
flamegraph -h

# Use sudo to run flamegraph
flamegraph --root -p $(pgrep zenoh-bridge-ro)

# Press Ctrl+c after 20-30sec
# a flamegraph.svg file is created

# Open flamegraph.svg in a browser and attach it here.

![flamegraph](https://github.com/user-attachments/assets/e1adf779-114a-4513-8117-9b9f8da2119b)

```

JEnoch, 2024-10-18:
Are you still experiencing this issue with version [`1.0.0-rc.2`](https://download.eclipse.org/zenoh//zenoh-plugin-ros2dds/1.0.0-rc.2/) ?

aosmw, 2024-10-21:
I am going to try this out today.  I have had our environments locked to "0.11.0-stable" for a while.

Side Note: My config was using a string for the "id:" field which is now rejected. 
```
Plugin "ros2dds" failed to start: Plugin `ros2dds` configuration error: unknown field `id` at zenoh-plugin-ros2dds/src/lib.rs:167
```
I am now commenting out the id field(allowing zenoh to generate its own random one), and putting the string I used to set id to into nodename.

I am writing this on the off chance that this may have been a trigger.

aosmw, 2024-12-03:
We are not experiencing the 100% CPU issue anymore with 1.0.0~rc.2-1.

JEnoch, 2024-12-04:
Thank you for your confirmation.

> I am writing this on the off chance that this may have been a trigger.

The `id` field was certainly not related, as it was only used in admin space, not at runtime for any routing.
I think the root cause was rather a temporary issue in zenoh 1.0.0-alpha, which was "work in progress".

Now that the `1.0.0` has officially been released (actually latest is `1.0.3`), I think you should switch to the latest release.

closed 2024-12-04 by JEnoch