eclipse-zenoh/zenoh-plugin-ros2dds #213: [Bug] - ros2 topic echo only works on the first call

firesurfer, 2024-08-14:
### Describe the bug

When running `ros2 topic echo /joint_states` (also happens for other topics) it works fine for the first call.
After closing `ros2 topic echo` with `CTRL+C` and running it again no topic output is shown. 
This happens also for other nodes. 

Bridge output:

```
2024-08-14T12:24:17.058238Z  INFO tokio-runtime-worker ThreadId(03) zenoh_plugin_ros2dds::routes_mgr: Route Service Server (ROS:/_ros2cli_2013878/get_type_description <-> Zenoh:_ros2cli_2013878/get_type_description) created
2024-08-14T12:24:17.058955Z  INFO tokio-runtime-worker ThreadId(03) zenoh_plugin_ros2dds: Node /_ros2cli_2013878 declares Publisher /parameter_events: rcl_interfaces/msg/ParameterEvent - Allowed
2024-08-14T12:24:17.058967Z  INFO tokio-runtime-worker ThreadId(03) zenoh_plugin_ros2dds: Node /_ros2cli_2013878 declares Publisher /rosout: rcl_interfaces/msg/Log - Allowed
2024-08-14T12:24:17.665454Z  INFO tokio-runtime-worker ThreadId(03) zenoh_plugin_ros2dds: Node /_ros2cli_2013878 declares Subscriber /joint_states: sensor_msgs/msg/JointState - Allowed
```

### To reproduce

1. Start the zenoh-bridge-ros2dds on the robot
2. Connect from another computer via `zenoh-bridge-ros2dds -e ...`
3. run `ros2 topic echo <topic>`
4. Close it with CTRL+C
5. run it again

### System info

-Ubuntu 22.04
-ROS2 iron
-zenoh-bridge-ros2dds v1.0.0-alpha.6

JEnoch, 2024-09-02:
Those logs looks normal: the bridge discovered the Subscriber on `/join_states` declared by the command `ros2 topic echo /join_states`.

I tried to reproduce using the `turtlebot3_gazebo` simulation on a Ubuntu 22.04 laptop with Iron and latest commit b993597.
I used those commands:
- `ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py`
- `./target/debug/zenoh-bridge-ros2dds`
- `ROS_DOMAIN_ID=1 ./target/debug/zenoh-bridge-ros2dds -l tcp/localhost:7448 -e tcp/localhost:7447`
- `ROS_DOMAIN_ID=1 ros2 topic echo /join_states sensor_msgs/msg/JointState` 
   then CTRL+C and run again

I didn't manage to reproduce the issue.

Could you please try again with latest commit of the `main` branch ?
And if the issue occurs again, please activate debug logs defining the `RUST_LOG=zenoh_plugin=debug` for the 2 bridges.

knorrrr, 2024-10-16:
@JEnoch 
Thank you for your production of the zenoh

I reproduced this bug.
This phenomenon occurs when stopping and restarting 'ros2 run demo_nodes_cpp talker' or 'ros2 topic echo /chatter', or when stopping and restarting Zenoh. 
[image]

Additionally, sometimes 'ros2 topic echo /chatter' echoes the output two or three times.
[image]

### **PC 01** running `ros2 run demo_nodes_cpp talker`
#### System Info
- Ubuntu 22.04
- zenoh-bridge-ros2dds v1.0.0-rc.1

<pre><font color="#26A269"><b>knorrrr@PC01</b></font>:<font color="#12488B"><b>~/</b></font>$ RUST_LOG=zenoh_plugin=debug zenoh-bridge-ros2dds 
<font color="#AAAAAA">2024-10-16T07:36:31.615440Z </font><font color="#12488B">DEBUG</font> main ThreadId(01) zenoh_plugin_trait::manager: Declared static plugin Id:ros2dds - Name:ros2dds
<font color="#AAAAAA">2024-10-16T07:36:31.615817Z </font><font color="#12488B">DEBUG</font> main ThreadId(01) zenoh_plugin_trait::manager::static_plugin: Plugin `ros2dds` started
<font color="#AAAAAA">2024-10-16T07:36:31.616299Z </font><font color="#12488B">DEBUG</font> tokio-runtime-worker ThreadId(22) zenoh_plugin_ros2dds: ROS2 plugin 1.0.0-rc.1
<font color="#AAAAAA">2024-10-16T07:36:31.616319Z </font><font color="#26A269"> INFO</font> tokio-runtime-worker ThreadId(22) zenoh_plugin_ros2dds: ROS2 plugin Config { namespace: &quot;/&quot;, nodename: &quot;zenoh_bridge_ros2dds&quot;, domain: 0, ros_localhost_only: false, allowance: None, pub_max_frequencies: [], transient_local_cache_multiplier: 10, queries_timeout: None, reliable_routes_blocking: true, pub_priorities: [], work_thread_num: 2, max_block_thread_num: 50, __required__: None, __path__: None }
<font color="#AAAAAA">2024-10-16T07:36:31.616479Z </font><font color="#12488B">DEBUG</font> tokio-runtime-worker ThreadId(22) zenoh_plugin_ros2dds: Create DDS Participant on domain 0 with CYCLONEDDS_URI=&apos;&apos;
<font color="#AAAAAA">2024-10-16T07:36:31.617777Z </font><font color="#12488B">DEBUG</font> tokio-runtime-worker ThreadId(22) zenoh_plugin_ros2dds: ROS2 plugin c36aa27fa7b5153d27189ed9b3242682 using DDS Participant 01101ee04fbcf2561ff4f7b9000001c1 created
<font color="#AAAAAA">2024-10-16T07:36:31.617849Z </font><font color="#12488B">DEBUG</font> tokio-runtime-worker ThreadId(22) zenoh_plugin_ros2dds: Declare admin space on @/c36aa27fa7b5153d27189ed9b3242682/ros2/**
<font color="#AAAAAA">2024-10-16T07:36:31.618250Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS Participant 011063bc4100871669cdf771000001c1)
<font color="#AAAAAA">2024-10-16T07:36:31.618354Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS Participant 01107117585a8f6c6b218b0d000001c1)
<font color="#AAAAAA">2024-10-16T07:36:31.619397Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 01107117585a8f6c6b218b0d00000403 from Participant 01107117585a8f6c6b218b0d000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619438Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 01107117585a8f6c6b218b0d00000603 from Participant 01107117585a8f6c6b218b0d000001c1 on rt/rosout with type rcl_interfaces::msg::dds_::Log_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619460Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 01107117585a8f6c6b218b0d00000703 from Participant 01107117585a8f6c6b218b0d000001c1 on rt/parameter_events with type rcl_interfaces::msg::dds_::ParameterEvent_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619476Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS subscription 01107117585a8f6c6b218b0d00000504 from Participant 01107117585a8f6c6b218b0d000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619503Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS subscription 01107117585a8f6c6b218b0d00000804 from Participant 01107117585a8f6c6b218b0d000001c1 on rt/chatter with type std_msgs::msg::dds_::String_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619520Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 011063bc4100871669cdf77100000403 from Participant 011063bc4100871669cdf771000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619542Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 011063bc4100871669cdf77100000603 from Participant 011063bc4100871669cdf771000001c1 on rt/rosout with type rcl_interfaces::msg::dds_::Log_ (keyless: true)
<font color="#AAAAAA">2024-10-16T07:36:31.619556Z </font><font color="#12488B">DEBUG</font> ThreadId(26) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 011063bc4100871669cdf77100000703 from Participant 011063bc4100871669cdf771000001c1 on rt/parameter_events with type rcl_interfaces::msg::dds_::ParameterEvent_ (keyless: true)
[... cut ...]
<font color="#AAAAAA">2024-10-16T07:36:59.843396Z </font><font color="#12488B">DEBUG</font> tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::route_subscriber: Route Subscriber (Zenoh:chatter -&gt; ROS:/chatter) now serving local nodes {&quot;/_ros2cli_75212&quot;}
<font color="#AAAAAA">2024-10-16T07:36:59.843481Z </font><font color="#12488B">DEBUG</font> tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::route_subscriber: Route Subscriber (Zenoh:chatter -&gt; ROS:/chatter) activate
</pre>

bigredfrog, 2026-01-29:
I took another run at this.

I was trying to go

zenoh_ros2dds  <-> lots of network abstractions <-> zenoh_ros2dds

After much nashing of teeth, I am now going

zenoh_ros2dds <-> network <-> zenoh router <-> zenoh_ros2dds

and so far I can't break it!