eclipse-zenoh/zenoh-plugin-ros2dds #219: [Bug] Transient local topics not working with second client

JMvanBruggen, 2024-08-21:
### Describe the bug

Second client does not receive transient local topics from a router. 

### To reproduce

1. Start a router: ``zenoh-bridge-ros2dds -m router``
2. Start a transient local publisher on the router side.
3. Start a client (on 2nd pc) ``zenoh-bridge-ros2dds -e tcp/ROBOT_IP:7447 -m client``
4. Start a second client (on 3rd pc) ``zenoh-bridge-ros2dds -e tcp/ROBOT_IP:7447 -m client``

One of both clients does not receive the transient local topic. (very problematic with /tf_static for example)

### System info

### System info

- Ubuntu 22:04
- ROS Humble
- dev/1.1.0 branch

muellerbernd, 2024-10-14:
I have noticed the same problem in my setup. Are there any solutions/hacks to get this working?

Morgantiz, 2025-07-04:
I have the same problem. Are there any updates? Have you found any solution?

JEnoch, 2025-08-29:
Hi there!
Thanks to @Morgantiz's contribution (#524) the bridge now uses the Advanvced Pub/Sub.
I hope this resolves the issue. Could you please test it with your use cases and let us know?