eclipse-zenoh/zenoh-plugin-ros2dds #249: Unable to call a ROS2 service from Zenoh

sumitpaulde, 2024-09-09:
### Describe the bug

I am using 2 physical machines.
 node1 host ROS2 service
 node2 host Zenoh client

Both running zenoh_bridge_ros2dds to establish communication between nodes.

ROS2 service requires a simple string as its service call. 
Service description
```
string taskname
---
string response
```
But I am trying to call the service with zenoh

```
@cdr
class Request:
       taskname: str

request_data = Request(taskname='ooo').serialize()
resource_name = "set_task_service"
for reply in session.get(resource_name, zenoh.Queue(), value=request_data):
    try:
        print(f"Received '{reply.ok.key_expr}': '{reply.ok.payload.decode('utf-8')}'")
    except:
        print(f"Received ERROR: '{reply.err.payload.decode('utf-8')}'")

```

the client request is not reaching to the ROS2 service server.

```
 rx-2 ThreadId(20) zenoh_plugin_ros2dds::route_service_srv: Route Service Server (ROS:/set_task_service <-> Zenoh:set_task_service): received invalid request: [7b, 22, 74, 61, 73, 6b, 6e, 61, 6d, 65, 22, 3a, 20, 22, 62, 6c, 61, 62, 6c, 61, 22, 7d]

``` 

The ros2 service code
```
#!/usr/bin/env python3
import rclpy
from rclpy.node import Node
from hare_robot_interfaces.srv import SetTask
from functools import partial

class SetTaskCleintNode(Node):
    
    def __init__(self):
        super().__init__("set_task_client")
        for x in range(10):
            self.call_set_task_service("image detection " + str(x))
    
        
    def call_set_task_service(self, value1):
        
        client = self.create_client(SetTask, "set_task_service")
        while not client.wait_for_service(1.0):
            self.get_logger().warn("Waiting for servers response!!")
        request = SetTask.Request()
        request.taskname = value1
        
        future = client.call_async(request)
     
        future.add_done_callback(partial(self.callback_call_set_task, value1 = value1))
    
    def callback_call_set_task(self, future, value1):
        try:
            response = future.result()
            self.get_logger().info("task send: " + value1 + " received " + response)
            
        except Exception as e:
            self.get_logger().error("Service call failed %r" % (e,))

                  
def main(args=None):
    rclpy.init(args= args)
    node = SetTaskCleintNode()
    rclpy.spin(node)
    rclpy.shutdown()
    
if __name__ == "__main__":
    main()
```

### To reproduce

to reproduce you can use the ROS2 service running on one machine and try to send the server the request from zenoh.

### System info

Ubuntu 22.04 arm 64
zenoh-bridge-ros2dds v0.11.0-dev-127-gb147cc7

TanJunKiat, 2024-09-16:
Are you using ROS2 humble?

JEnoch, 2024-09-30:
I think #277 might fix this.
Could you please test again with latest commit (1aaa0c6e9145f11a903b9cf9b28ee9fe92008ab4) ?

sumitpaulde, 2024-10-15:
Hi @JEnoch, 
Thanks a lot for replying!

I have tried the latest commit; the problem is the previous error is gone, but there is no communication happening between the zenoh client and ros2 Server. Earlier, after sending the service call from the zenoh client, the zenoh_bridge_ros2dds used to receive data and show some error logs on the terminal; now, even that is not happening.

sample zenoh client code:

```
import sys
import time
import argparse
import json
import zenoh
from zenoh import config, QueryTarget
from pycdr import cdr

conf = zenoh.Config()
selector = "set_task_service"
target= QueryTarget.ALL()

@cdr
class Request:
       taskname: str

def main():
    zenoh.init_logger()

    print("Opening session...")
    session = zenoh.open(conf)

# here we tried with JSON/ Request Type/ simple string but nothing worked.

# update the JSON data according to your need
#    json_data = {
#        "taskname": "abcabc"}
#    req = json.dumps(json_data).encode('utf-8')

#  req = Request(taskname='abcabc').serialize()

    req = ("abcabc").encode("utf-8")
    print("Sending Query '{}'...".format(selector))
    replies = session.get(selector, zenoh.Queue(), target=target, value=req, consolidation=zenoh.QueryConsolidation.NONE())
    for reply in replies.receiver:
        try:
            print(">> Received ('{}': '{}')"
                .format(reply.ok.key_expr, reply.ok.payload.decode("utf-8")))
        except:
            print(">> Received (ERROR: '{}')"
                .format(reply.err.payload.decode("utf-8")))

    session.close()

main()

```
## On the ros2 service node
zenoh_bridge_ros2dds 
## On the zenoh client side
zenoh_bridge_ros2dds -e tcp/192.168.1.72:7447

## Output from the zenoh_bridge_ros2dds
```
2024-10-15T15:13:17.082287Z  INFO tokio-runtime-worker ThreadId(03) zenoh_plugin_ros2dds::routes_mgr: Route Service Client (ROS:/set_task_server/list_parameters <-> Zenoh:set_task_server/list_parameters) created
2024-10-15T15:13:17.176353Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_server/set_parameters: rcl_interfaces/srv/SetParameters - Allowed
2024-10-15T15:13:17.177230Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_server/describe_parameters: rcl_interfaces/srv/DescribeParameters - Allowed
2024-10-15T15:13:17.177824Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_service: hare_robot_interfaces/srv/SetTask - Allowed
2024-10-15T15:13:17.178387Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_server/list_parameters: rcl_interfaces/srv/ListParameters - Allowed
2024-10-15T15:13:17.178771Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_server/set_parameters_atomically: rcl_interfaces/srv/SetParametersAtomically - Allowed
2024-10-15T15:13:17.179429Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_server/get_parameters: rcl_interfaces/srv/GetParameters - Allowed
2024-10-15T15:13:17.179787Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_server/get_parameter_types: rcl_interfaces/srv/GetParameterTypes - Allowed

```
Kind Regards

JEnoch, 2024-10-18:
Can you make sure you're using the same version for the bridge and for `zenoh-python`.
If still not working, please run your client with `RUST_LOG=debug` environment variable and provide the logs.

sumitpaulde, 2024-10-18:
Yeah, I can confirm that I am using the same version, as I built the Docker container on one device and used the same Docker container on the second machine.

So the ```zenoh-python``` and ```zenoh_bridge_ros2dds``` are same on the both nodes, both nodes are physically with same coniguration as well.

sumitpaulde, 2024-10-22:
Hi @JEnoch, you are right there is some version mismatching I understood from the log, but I do not understand which versions it is speaking about.

```
2024-10-22T09:51:02.912484Z  INFO tokio-runtime-worker ThreadId(05) zenoh_plugin_ros2dds: Node /zenoh_bridge_ros2dds declares Service Client /set_task_service: hare_robot_interfaces/srv/SetTask - Allowed
2024-10-22T09:51:02.912671Z DEBUG tokio-runtime-worker ThreadId(05) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/set_task_service <-> Zenoh:set_task_service): now serving local nodes {"/zenoh_bridge_ros2dds"}
2024-10-22T09:51:02.912845Z DEBUG tokio-runtime-worker ThreadId(05) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/set_task_service <-> Zenoh:set_task_service): announce via token @/f9cd6f8e79e88d43aa14144ed271a4ed/@ros2_lv/SC/set_task_service/hare_robot_interfaces§srv§SetTask
2024-10-22T09:51:02.913058Z DEBUG tokio-runtime-worker ThreadId(05) zenoh::net::routing::dispatcher::token: Face{1, f9cd6f8e79e88d43aa14144ed271a4ed} Declare token 36 (@/f9cd6f8e79e88d43aa14144ed271a4ed/@ros2_lv/SC/set_task_service/hare_robot_interfaces§srv§SetTask)
2024-10-22T09:51:02.913279Z DEBUG tokio-runtime-worker ThreadId(05) zenoh::net::routing::dispatcher::resource: Register resource @/f9cd6f8e79e88d43aa14144ed271a4ed/@ros2_lv/SC/set_task_service/hare_robot_interfaces§srv§SetTask
2024-10-22T09:51:29.417185Z DEBUG                acc-0 ThreadId(07) zenoh_link_tcp::unicast: Accepted TCP connection on [::ffff:192.168.1.72]:7447: [::ffff:192.168.1.49]:44038
2024-10-22T09:51:29.417942Z DEBUG                acc-0 ThreadId(07) zenoh_transport::unicast::establishment::accept: Rejecting InitSyn on tcp/[::ffff:192.168.1.72]:7447 => tcp/[::ffff:192.168.1.49]:44038 because of unsupported Zenoh protocol version (expected: 9, received: 8) from: 7f40d4d0775a530447beb7c29002e6ab at /root/.cargo/git/checkouts/zenoh-cc237f2570fab813/e79c800/io/zenoh-transport/src/unicast/establishment/accept.rs:186.
2024-10-22T09:51:30.439713Z DEBUG                acc-0 ThreadId(07) zenoh_link_tcp::unicast: Accepted TCP connection on [::ffff:192.168.1.72]:7447: [::ffff:192.168.1.49]:44050
2024-10-22T09:51:30.440309Z DEBUG                acc-0 ThreadId(07) zenoh_transport::unicast::establishment::accept: Rejecting InitSyn on tcp/[::ffff:192.168.1.72]:7447 => tcp/[::ffff:192.168.1.49]:44050 because of unsupported Zenoh protocol version (expected: 9, received: 8) from: 7f40d4d0775a530447beb7c29002e6ab at /root/.cargo/git/checkouts/zenoh-cc237f2570fab813/e79c800/io/zenoh-transport/src/unicast/establishment/accept.rs:186.
2024-10-22T09:51:32.385476Z DEBUG                acc-0 ThreadId(07) zenoh_link_tcp::unicast: Accepted TCP connection on [::ffff:192.168.1.72]:7447: [::ffff:192.168.1.49]:44908
2024-10-22T09:51:32.386135Z DEBUG                acc-0 ThreadId(07) zenoh_transport::unicast::establishment::accept: Rejecting InitSyn on tcp/[::ffff:192.168.1.72]:7447 => tcp/[::ffff:192.168.1.49]:44908 because of unsupported Zenoh protocol version (expected: 9, received: 8) from: 7f40d4d0775a530447beb7c29002e6ab at /root/.cargo/git/checkouts/zenoh-cc237f2570fab813/e79c800/io/zenoh-transport/src/unicast/establishment/accept.rs:186.
```

sumitpaulde, 2024-10-24:
Hi @JEnoch I have changed the Zenoh version of the client node, so the previous version mismatch error is not visible now but I am getting another error now. Can you please suggest me what am I doing wrong?

```
2024-10-24T17:43:38.453993Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:377
2024-10-24T17:43:38.454035Z  WARN                 rx-0 ThreadId(11) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/set_task_service <-> Zenoh:set_task_service): received NO reply for request (ed8e9a3363c3d394,0) - cannot reply to client, it will hang until timeout
2024-10-24T17:43:38.457976Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Route query Face{2, 4fd10ecb817c162b7256e87b34892c4e}:407 for res set_task_service
2024-10-24T17:43:38.458133Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Received final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:406 from Face{1, 124714bcc7c5947c508b1a4cd6c330ba}
2024-10-24T17:43:38.458197Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:406
2024-10-24T17:43:38.462300Z DEBUG ThreadId(13) zenoh::net::routing::dispatcher::queries: Route query Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:380 for res set_task_service
2024-10-24T17:43:38.462532Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Received final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:379 from Face{2, 4fd10ecb817c162b7256e87b34892c4e}
2024-10-24T17:43:38.462609Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:378
2024-10-24T17:43:38.462653Z  WARN                 rx-0 ThreadId(11) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/set_task_service <-> Zenoh:set_task_service): received NO reply for request (ed8e9a3363c3d394,0) - cannot reply to client, it will hang until timeout
2024-10-24T17:43:38.463766Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Route query Face{2, 4fd10ecb817c162b7256e87b34892c4e}:408 for res set_task_service
2024-10-24T17:43:38.463926Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Received final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:407 from Face{1, 124714bcc7c5947c508b1a4cd6c330ba}
2024-10-24T17:43:38.463981Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:407
2024-10-24T17:43:38.467565Z DEBUG ThreadId(13) zenoh::net::routing::dispatcher::queries: Route query Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:381 for res set_task_service
2024-10-24T17:43:38.468173Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Received final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:380 from Face{2, 4fd10ecb817c162b7256e87b34892c4e}
2024-10-24T17:43:38.468263Z DEBUG                 rx-0 ThreadId(11) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:379
2024-10-24T17:43:38.468306Z  WARN                 rx-0 ThreadId(11) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/set_task_service <-> Zenoh:set_task_service): received NO reply for request (ed8e9a3363c3d394,0) - cannot reply to client, it will hang until timeout
2024-10-24T17:43:38.469356Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Route query Face{2, 4fd10ecb817c162b7256e87b34892c4e}:409 for res set_task_service
2024-10-24T17:43:38.469502Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Received final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:408 from Face{1, 124714bcc7c5947c508b1a4cd6c330ba}
2024-10-24T17:43:38.469551Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:408
2024-10-24T17:43:38.478391Z DEBUG ThreadId(13) zenoh::net::routing::dispatcher::queries: Route query Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:382 for res set_task_service
2024-10-24T17:43:38.478829Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Received final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:381 from Face{2, 4fd10ecb817c162b7256e87b34892c4e}
2024-10-24T17:43:38.478935Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{1, 124714bcc7c5947c508b1a4cd6c330ba}:380
2024-10-24T17:43:38.479023Z  WARN                 rx-1 ThreadId(12) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/set_task_service <-> Zenoh:set_task_service): received NO reply for request (ed8e9a3363c3d394,0) - cannot reply to client, it will hang until timeout
2024-10-24T17:43:38.479189Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Route query Face{2, 4fd10ecb817c162b7256e87b34892c4e}:410 for res set_task_service
2024-10-24T17:43:38.479308Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Received final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:409 from Face{1, 124714bcc7c5947c508b1a4cd6c330ba}
2024-10-24T17:43:38.479407Z DEBUG                 rx-1 ThreadId(12) zenoh::net::routing::dispatcher::queries: Propagate final reply Face{2, 4fd10ecb817c162b7256e87b34892c4e}:409

```

sumitpaulde, 2024-11-19:
Hi @JEnoch, I know you are busy with lots of things. I would like to ask if you can invest a few moments on this topic and guide me.

JEnoch, 2024-11-25:
This log:
`Route Service Client (...): received NO reply for request ...`
indicates that the bridge routed a request from DDS to Zenoh, but no Zenoh application (bridge or zenoh-python script) replied to this request.

Which scenario are you testing ?
It no longer seems to be a Zenoh client calling a ROS 2 Service via the bridge, but rather the opposite: a ROS 2 client calling a Zenoh service ?

edgimar, 2025-02-26:
FWIW, I was looking for a minimum working example that works with the ros2dds bridge, and a natural candidate would be the [zenoh-rust-teleop](https://github.com/eclipse-zenoh/zenoh-demos/tree/main/ROS2/zenoh-rust-teleop) executable, but this depends on Zenoh 0.10.0-rc, which ends up having the same protocol mismatch issue.  Is there an example which is kept in sync with the latest Zenoh version?  If not, maybe it would be worthwhile to update these demos?

TanJunKiat, 2025-02-26:
@edgimar 

I made a PR for zenoh 1.0.0 to update the examples for `zenoh-bridge-ros2dds`, the `example` branch.

PR: https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/317

Feel free to refer to this PR to get a somewhat working copy for 1.0.0.

edgimar, 2025-02-28:
FWIW, I also updated the `ros2-teleop` example (to use 1.2.1) here: https://github.com/edgimar/zenoh-demos/blob/main/ROS2/zenoh-rust-teleop/src/bin/ros2-teleop.rs

Although it runs, it seems unable to control a `turtlesim_node` turtle. :disappointed:

sumitpaulde, 2025-03-09:
Hi, Thanks a lot for your update. I tried to run the ROS2 Service from Zenoh Client and vice versa, but I'm still having a communication issue. Do you have any suggestions for me? I tried to run the service also with the cyclone dds rmw.

TanJunKiat, 2025-03-09:
> Hi, Thanks a lot for your update. I tried to run the ROS2 Service from Zenoh Client and vice versa, but I'm still having a communication issue. Do you have any suggestions for me? I tried to run the service also with the cyclone dds rmw.

Do you have more information? 
Are you running the zenoh client and ros2 service server on the same environment? 

Did you turn on zenoh bridge?

What is the error message you see when you run your script?