eclipse-zenoh/zenoh-plugin-ros2dds #271: Zenoh-bridge-ros2dds not working in Jazzy.

RishikesavanRamesh, 2024-09-23:
### Describe the release item

[image]
The above screenshot is of Jazzy version of a docker container, in which i tried to service call the /add_two_ints service, using zenoh-bridge-ros2dds and curl. It is not working as expected. But when I run the container with humble base image, as in the below screenshot it is working. How can I make it work in Jazzy ?
[image]

JEnoch, 2024-10-18:
In your Jazzy screenshot the bridge isn't discovering anything over DDS.  I just did the same test than you on Jazzy with `rmw_cyclonedds_cpp` and it works for me.  
Aren't you using `rmw_cyclonedds_cpp` ?

evshary, 2024-10-23:
@RishikesavanRamesh it seems like you enable the `ROS_LOCALHOST_ONLY`. The mechanism of `ROS_LOCALHOST_ONLY` changes after ROS 2 Iron (BTW, It's called `ROS_AUTOMATIC_DISCOVERY_RANGE` now). I will make a patch for this. For the time being, you can disable that to make it work.

RishikesavanRamesh, 2024-10-23:
> In your Jazzy screenshot the bridge isn't discovering anything over DDS. I just did the same test than you on Jazzy with `rmw_cyclonedds_cpp` and it works for me. Aren't you using `rmw_cyclonedds_cpp` ?

@JEnoch Yes, I am using `rmw_cyclonedds_cpp`.

RishikesavanRamesh, 2024-10-23:
> @RishikesavanRamesh it seems like you enable the `ROS_LOCALHOST_ONLY`. The mechanism of `ROS_LOCALHOST_ONLY` is changed after ROS 2 Iron (BTW, It's called `ROS_AUTOMATIC_DISCOVERY_RANGE` now). The CycloneDDS configuration is different in that case.
> 
> I will make a patch for this. For the time being, you can disable that to make it work.

Thank you @evshary . This will do for now.

closed 2024-10-23 by RishikesavanRamesh

evshary, 2024-10-23:
The PR should fix this issue. FYI
https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/pull/310