eclipse-zenoh/zenoh-plugin-ros2dds #273: [Bug] Moveit send_goal action receiving invalid request

miltzhaw, 2024-09-26:
### Describe the bug

Running rviz + moveit on a container  in a kubernetes cluster and communicate over the zenoh bridge to a container running on a physical robot.
Able to get topics, actions and robot model. Able to plan a movement of the robotic arm succesfully. But when executing the movement the action is not working properly. 

On the side where I send the action command (via moveit and rviz) I see the following warning:

`2024-09-26T13:50:03.052308Z  WARN                 rx-1 ThreadId(15) zenoh_plugin_ros2dds::route_service_cli: Route Service Client (ROS:/summit/arm_controller/follow_joint_trajectory/_action/send_goal <-> Zenoh:bot1/summit/arm_controller/follow_joint_trajectory/_action/send_goal): received NO reply for request (587017f51973ff7b,1) - cannot reply to client, it will hang until timeout`

 On the receiving end (the container on the robot), I see the following warning:

`-09-26T13:50:02.433624Z  WARN                 rx-0 ThreadId(15) zenoh_plugin_ros2dds::route_service_srv: Route Service Server (ROS:/summit/arm_controller/follow_joint_trajectory/_action/send_goal <-> Zenoh:bot1/summit/arm_controller/follow_joint_trajectory/_action/send_goal): received invalid request: [90, 24, 53, 36, 7b, e5, 90, 1b, 9.....`

From my understanding this action is receiving an invalid message and therefore blocked. The question is why is it getting an invalid request?
Note that other actions are working, for instance the action for the gripper works in the same setup or also Nav2 was tested. 

Tried to increase the queries_timeout for the actions in the configuration files for the Zenoh ros2dds bridge on both sides, with no success.

### To reproduce

1. Start ros2-dds-bridge on the robot with almost default configuration to connect to a Zenoh router in client mode. 
2. Start ros2-dds-bridge on the container in the cloud with almost default configuration to connect to a Zenoh router in client mode.
3. Start rviz with moveit on the cloud container plan a movement of the robotic arm and then try to execute

### System info

Robot with ROS2 humble container and ros2ddsbridge version 1.0.0~beta.3-1 (tested also previous versions)
Container on Kubernetes cluster with ROS2 humble and ros2ddsbridge version 1.0.0~beta.3-1 (tested also previous versions)
Zenoh router version 1.0.0-beta.3 (tested also previous versions)

JEnoch, 2024-09-30:
I think #277 might fix this.
Could you please test again with latest commit (1aaa0c6e9145f11a903b9cf9b28ee9fe92008ab4) ?

miltzhaw, 2024-10-01:
It solved indeed the issue! Thank you for reacting so quickly.

I used the last nightly build of the zenoh-ros2dds-bridge on both sides and it works. 

I also tested nav2 actions and it also works fine. However, after a couple of nav2 goals I get a lot of warning messages like the following ones:

`2024-10-01T10:24:21.710278Z  WARN                 rx-1 ThreadId(13) zenoh_plugin_ros2dds::route_subscriber: Route Subscriber (Zenoh:bot1/summit/bond -> ROS:/summit/bond): can't route message; sertype lookup failed (Bad Parameter)`

JEnoch, 2024-10-01:
Thanks for your confirmation. I'm closing this issue then.

> However, after a couple of nav2 goals I get a lot of warning messages like the following ones:
> `2024-10-01T10:24:21.710278Z  WARN                 rx-1 ThreadId(13) [...]

It's the same issue than #278. We're investigating this.

closed 2024-10-01 by JEnoch