eclipse-zenoh/zenoh-plugin-ros2dds #295: [Bug]

tomkimsour, 2024-10-14:
### Describe the bug

I am running ubuntu 22 and ROS2 Humble, I want to spin 2 zenoh-bridge, each having a different ROS_DOMAIN_ID.
By default they are set as router and I would like to add the rule so router can autoconnect to each other using the following scouting config.

```
  scouting: {
    /// The UDP multicast scouting configuration.
    multicast: {
      /// Whether multicast scouting is enabled or not
      enabled: true,
      /// The socket which should be used for multicast scouting
      address: "224.0.0.224:7446",
      /// The network interface which should be used for multicast scouting
      interface: "auto", // If not set or set to "auto" the interface if picked automatically
      /// Which type of Zenoh instances to automatically establish sessions with upon discovery on UDP multicast.
      /// Accepts a single value or different values for router, peer and client.
      /// Each value is bit-or-like combinations of "peer", "router" and "client".
      autoconnect: { router: 'router', peer: "router|peer" },
      /// Whether or not to listen for scout messages on UDP multicast and reply to them.
      listen: true,
    },
    /// The gossip scouting configuration.
    gossip: {
      /// Whether gossip scouting is enabled or not
      enabled: true,
      /// When true, gossip scouting information is propagated multiple hops to all nodes in the local network.
      /// When false, gossip scouting information is only propagated to the next hop.
      /// Activating multihop gossip implies more scouting traffic and a lower scalability.
      /// It mostly makes sense when using "linkstate" routing mode where all nodes in the subsystem don't have
      /// direct connectivity with each other.
      multihop: false,
      /// Which type of Zenoh instances to automatically establish sessions with upon discovery on gossip.
      /// Accepts a single value or different values for router, peer and client.
      /// Each value is bit-or-like combinations of "peer", "router" and "client".
      autoconnect: { router: "router", peer: "router|peer" },
    },
  },
```

The following error appear saying the json is expecting a list
```
2024-10-14T13:44:30.178539Z  INFO main ThreadId(01) zenoh_bridge_ros2dds: zenoh-bridge-ros2dds v1.0.0-beta.4
thread 'main' panicked at zenoh-bridge-ros2dds/src/zenoh_args.rs:77:51:
called `Result::unwrap()` on an `Err` value: JSON error: invalid type: string "router", expected a list of whatami variants ('router', 'peer', 'client') at /home/runner/.cargo/registry/src/index.crates.io-6f17d22bba15001f/zenoh-config-1.0.0-beta.4/src/lib.rs:778.
stack backtrace:
   0:     0x59a813fb20ac - std::backtrace_rs::backtrace::libunwind::trace::ha637c64ce894333a
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/../../backtrace/src/backtrace/libunwind.rs:104:5
   1:     0x59a813fb20ac - std::backtrace_rs::backtrace::trace_unsynchronized::h47f62dea28e0c88d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x59a813fb20ac - std::sys_common::backtrace::_print_fmt::h9eef0abe20ede486
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x59a813fb20ac - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::hed7f999df88cc644
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:44:22
   4:     0x59a813d3ca30 - core::fmt::rt::Argument::fmt::h1539a9308b8d058d
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/fmt/rt.rs:142:9
   5:     0x59a813d3ca30 - core::fmt::write::h3a39390d8560d9c9
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/fmt/mod.rs:1120:17
   6:     0x59a813f7e972 - std::io::Write::write_fmt::h5fc9997dfe05f882
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/io/mod.rs:1762:15
   7:     0x59a813fb3c5e - std::sys_common::backtrace::_print::h894006fb5c6f3d45
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:47:5
   8:     0x59a813fb3c5e - std::sys_common::backtrace::print::h23a2d212c6fff936
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:34:9
   9:     0x59a813fb33d0 - std::panicking::default_hook::{{closure}}::h8a1d2ee00185001a
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:272:22
  10:     0x59a813fb4592 - std::panicking::default_hook::h6038f2eba384e475
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:292:9
  11:     0x59a813fb4592 - std::panicking::rust_panic_with_hook::h2b5517d590cab22e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:779:13
  12:     0x59a813fb3fbc - std::panicking::begin_panic_handler::{{closure}}::h233112c06e0ef43e
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:657:13
  13:     0x59a813fb3f16 - std::sys_common::backtrace::__rust_end_short_backtrace::h6e893f24d7ebbff8
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/sys_common/backtrace.rs:170:18
  14:     0x59a813fb3f01 - rust_begin_unwind
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/std/src/panicking.rs:645:5
  15:     0x59a813bcfa04 - core::panicking::panic_fmt::hbf0e066aabfa482c
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/panicking.rs:72:14
  16:     0x59a813bcff72 - core::result::unwrap_failed::hddb4fea594200c52
                               at /rustc/82e1608dfa6e0b5569232559e3d385fea5a93112/library/core/src/result.rs:1653:5
  17:     0x59a813c930b7 - zenoh_bridge_ros2dds::parse_args::h0c528a8edf61f891
  18:     0x59a813ca12e7 - zenoh_bridge_ros2dds::main::{{closure}}::h40f87851c276dcc0
  19:     0x59a813c9f16f - zenoh_bridge_ros2dds::main::h0bf36f20b2fd1b30
  20:     0x59a813c3d6e3 - std::sys_common::backtrace::__rust_begin_short_backtrace::h9b7c0d9b1b980ace
  21:     0x59a813c3e0c9 - std::rt::lang_start::hed384f925d67473f
  22:     0x7e6752748d90 - __libc_start_call_main
                               at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  23:     0x7e6752748e40 - __libc_start_main_impl
                               at ./csu/../csu/libc-start.c:392:3
  24:     0x59a813c2299e - _start
  25:                0x0 - <unknown>
Aborted (core dumped)```

### To reproduce

- open one terminal and launch `zenoh-bridge-ros2dds`
- open a second terminal run `export ROS_DOMAIN_ID=2`
- copy the default file under /etc/zenoh-bridge-ros2dds/conf.JSON5 to your /tmp
- modify the copied file by removing the comment of the scouting section
- run `zenoh-bridge-ros2dds -c /tmp/conf.JSON5`

### System info

Ubuntu 22.04, ROS2 Humble

gabrik, 2024-10-14:
Hi @tomkimsour seem that you are using an invalid configuration.

Your configuration should look like:

```json5

scouting: {
    /// The UDP multicast scouting configuration.
    multicast: {
      /// Whether multicast scouting is enabled or not
      enabled: true,
      /// The socket which should be used for multicast scouting
      address: "224.0.0.224:7446",
      /// The network interface which should be used for multicast scouting
      interface: "auto", // If not set or set to "auto" the interface if picked automatically
      /// Which type of Zenoh instances to automatically establish sessions with upon discovery on UDP multicast.
      /// Accepts a single value or different values for router, peer and client.
      /// Each value is bit-or-like combinations of "peer", "router" and "client".
      autoconnect: { router: ["router"], peer: ["router","peer"] }, // This is the correct format
      /// Whether or not to listen for scout messages on UDP multicast and reply to them.
      listen: true,
    },
    /// The gossip scouting configuration.
    gossip: {
      /// Whether gossip scouting is enabled or not
      enabled: true,
      /// When true, gossip scouting information is propagated multiple hops to all nodes in the local network.
      /// When false, gossip scouting information is only propagated to the next hop.
      /// Activating multihop gossip implies more scouting traffic and a lower scalability.
      /// It mostly makes sense when using "linkstate" routing mode where all nodes in the subsystem don't have
      /// direct connectivity with each other.
      multihop: false,
      /// Which type of Zenoh instances to automatically establish sessions with upon discovery on gossip.
      /// Accepts a single value or different values for router, peer and client.
      /// Each value is bit-or-like combinations of "peer", "router" and "client".
      autoconnect: { router: "router", peer: "router|peer" },
    },
  },
```

tomkimsour, 2024-10-14:
> Hi @tomkimsour seem that you are using an invalid configuration.
Hi, thanks for your quick response !
I ended up figuring it out while checking the config from zenoh repo. I opened a PR regarding the default config of zenoh-plugin-ros2dds.

gabrik, 2024-10-14:
Thank you for spotting this, and for your PR, we are going to merge it once the CI passes.

closed 2024-10-14 by gabrik