eclipse-zenoh/zenoh-plugin-ros2dds #305: [Bug] Unable to work with two bridges on different namespace

iamkashish, 2024-10-21:
### Describe the bug

If I start one bridge with a namespace "/left" and then the other with namespace "/right", the topics from left to right or vice-versa are created. But the publisher get lost, meaning: ros2 topic echo /right/<topic-name> doesn't show anything.
But if I use one bridge with a namespace "/left" and then other with no namespace. It works fine.
Maybe there is some silly mistake, but I tried various ways.

### To reproduce

1. Start `zenoh-bridge-ros2dds --namespace left -e tcp/<left-ip-address>:7447` in one machine.
2. Start `zenoh-bridge-ros2dds --namespace right -e tcp/<right-ip-address>:7447` in another machine.
3. Start `ros2 run demo_nodes_cpp talker` in left machine 
4. Do `ros2 echo /left/chatter` in right machine

### System info

- Platform Ubuntu 24.04

gabrik, 2024-10-22:
Hi @iamkashish, can you please share the logs of the two bridges when setting `RUST_LOG=z=debug` as environment variable?

iamkashish, 2024-11-04:
Hey @gabrik ,
I hope you are doing amazing. I'll glad if I can get some insights from you, whether I'm doing something wrong while deploying the bridge or is it really a bug?

iamkashish, 2024-11-18:
Hey @gabrik ,
I hope you are doing well. Looking forward to your reply.

evshary, 2024-11-19:
Hi @iamkashish
I think the behavior can be expected. Now zenoh-bridge-ros2dds doesn't support two hosts having different namespaces communicate with each other.
While you try `ros2 echo /left/chatter` on your right machine, the bridge is actually trying to subscribe `/right/left/chatter` in Zenoh network, which is not correct.

The only way to exchange the data now is to have both sides using same namespace or one side without namespace.
This is the limitation for the current implementation.

I think your requirement is indeed useful, but we need to think of a better way for bridge to identify whether `/left` is the namespace instead of part of topic.

RishikesavanRamesh, 2024-12-30:
Can we expect this feature anytime soon?

evshary, 2024-12-31:
> Can we expect this feature anytime soon?

As far as I know, this will not happen in the near future. However, the priority might change if the community makes more requests.

murph12F, 2025-01-06:
had the same problem, spent hours on it before finding this, i think this would be hella helpful, how come is not a default behavior to be able to have different entities with different namespaces ?  how complex would it be to change this? maybe if you point me in the right direction i can take some time to contribute for this. 

Hope this gets introduced soon, i was relying on the possibility to have different nodes have their namespace kinda like /drone/<id> namespace for each uav (each one on a separate network).