eclipse-zenoh/zenoh-plugin-ros2dds #311: Support `ROS_AUTOMATIC_DISCOVERY_RANGE` and `ROS_AUTOMATIC_DISCOVERY_RANGE`

evshary, 2024-10-23:
### Describe the feature

Based on the description on the description of Dynamic Discovery, we might need to make some apropriate changes.
https://docs.ros.org/en/rolling/Tutorials/Advanced/Improved-Dynamic-Discovery.html#id4

1. `ROS_LOCALHOST_ONLY` will be deprecated in the future, but to ensure the backward compatibility, `--ros-localhost-only` should be kept and perhaps show warning to users if they are using ROS 2 Iron or later.
2. Now we have both `ROS_AUTOMATIC_DISCOVERY_RANGE` and `ROS_AUTOMATIC_DISCOVERY_RANGE` environmental variables and will affect the DDS configuration. We might also need to support these two arguments `--ros-automatic-discovery-range` and `--ros-static-peers`.

@JEnoch I would like to know your thoughts about this.

evshary, 2024-10-23:
We can refer to [rmw_cyclonedds_cpp](https://github.com/ros2/rmw_cyclonedds/blob/c9e7001e6bf5373bdf1931535354b52eeddb2053/rmw_cyclonedds_cpp/src/rmw_node.cpp#L1134) for the DDS configuration.

evshary, 2024-11-08:
@JEnoch, since we agreed on the proposal during the meeting, I will start implementing it.

JEnoch, 2024-11-25:
Implemented in #334 .