eclipse-zenoh/zenoh-plugin-ros2dds #331: [Bug] Connection between zenoh-bridge-ros2dds and regular ros2 zenoh node

KenN7, 2024-11-08:
### Describe the bug

A ros2 node with rmw_zenoh cannot to connect to the router of the zenoh-bridge-ros2dds.
```
ERROR net-0 ThreadId(03) zenoh_transport::unicast::establishment::open: 
Received a close message (reason INVALID) in response to an InitSyn
```

### To reproduce

Launch the docker image `eclipse/zenoh-bridge-ros2dds:latest`
Then launch a ros2 node with zenoh as rmw (e.g. https://github.com/ZettaScaleLabs/roscon2024_workshop/blob/main/docker/Dockerfile).
The ros2 node will not connect to the router of the bridge.
This is probably due to a problem of zenoh version between the two, but I have no idea how to sync the version between the 2 software.
the rwm_zenoh has only "rolling" version, whereas the bridge feature versions.

### System info

- Nvidia jetson platform, bridge and nodes running in docker containers.

Mallets, 2024-11-08:
`rmw_zenoh` is still using zenoh `0.11.0` while `eclipse/zenoh-bridge-ros2dds:latest` is already zenoh `1.0.2`.
With `rmw_zenoh` you should use `0.11.0` bridge. 

In addition, the `rmw_zenoh` porting to zenoh 1.0.2 is happening [here](https://github.com/ros2/rmw_zenoh/pull/276). 
Any additional testing is also very welcome.

JEnoch, 2024-11-08:
Still `zenoh-bridge-ros2dds` is not compatible with `rmw_zenoh`, thus it won't work even if underlying Zenoh versions are the same.

The reason being that `rmw_zenoh` is using a different mapping of ROS msg/srv/actions to Zenoh key expressions than the `zenoh-bridge-ros2dds`. The key expressions for the Liveliness tokens used for discovery are different as well.

We have no short term plan to make the `zenoh-bridge-ros2dds` compatible with `rmw_zenoh`, as this is a significant work.

closed 2024-11-08 by JEnoch