eclipse-zenoh/zenoh-plugin-ros2dds #337: [Bug] zenoh_bridge_ros2dds doesn't announce a subscriber

yuma-m, 2024-11-19:
### Describe the bug

We use `zenoh_bridge_ros2dds` standalone executable to bridge ROS messages between two hosts (host A and host B) with the commands below. Host A and B are connected with a wired network.

- Host A: `zenoh_bridge_ros2dds -l tcp/0.0.0.0:7447 -c <Path to the config below>`
- Host B: `zenoh_bridge_ros2dds -e tcp/<Host A's IP>:7447`

Our config file is like below.

```
{
  plugins: {
    ros2dds: {
      allow: {
        publishers: ["/tf", "/tf_static", "topic_a", "topic_b", ... ],  # we have about 10 topics
        subscribers: ["/diagnostics", "topic_c", "topic_d", ...],  # about 20 topics
        service_servers: ["/some_service", ...],  # about 10 services
        service_clients: ["/other_service_.+/.*",  #  1 service with wildcard
        action_servers: [],
        action_clients: [],
      },
    },
  },
}
```

The `diagnostic_aggregator` node on host A subscribes to the `/diagnostics` topic, and multiple nodes on both host A and B publishes messages to the `/diagnostics` topic.

With this setup, I found an issue that Zenoh sometimes doesn't transfer the `/diagnostics` topic from host B to host A even if there's a subscriber on host A. When the issue happens, `zenoh_bridge_ros2dds` on host B doesn't subscribe to the `/diagnostics` topic, and the remote bridge (on the host A) doesn't announce a subscriber for the `/diagnostics` topic. If I manually run `ros2 topic echo /diagnostics` on host A, `zenoh_bridge_ros2dds` starts to transfer the messages from host B to A, but if I stop the `ros2 topic echo` command, it stops transferring.

This issue randomly happens when the ROS nodes on host A are launched. I also found the same issue happened on topics other than the `/diagnostics` (`topic_c` and `topic_d` in the configuration above) randomly, so I suspect Zenoh can sometimes fail to find a subscriber to specific topics depending on the launching timing.

Could you kindly guide me how to investigate this issue further or mitigate it?

### To reproduce

I didn't find a simple way to reproduce the issue, but the issue sometimes happens in my environment with the configurations above. I can provide more information about the setup, logs, and so on, if you need.

### System info

- Platform
  - Host A: Ubuntu 22.04 amd64
  - Host B: Ubuntu 20.04 arm64
- ROS2 version: Humble
- zenoh_bridge_ros2dds version: 1.0.1

yuma-m, 2024-11-20:
Hi @evshary, Thank you for your quick response. Let me clarify our configuration.

> I suppose you also have `/diagnostics` here, but just want to confirm with you.

Actually, I have `/diagnostics` topic only in `subscribers`, because I use this configuration only for the `zenoh_bridge_ros2dds` on host A. `zenoh_bridge_ros2dds` on host B allows any topic.

> Do you build the ROS 2 Humble by yourself? I guess ROS 2 Humble doesn't support Ubuntu 20.04.

Sorry, this is a bit wrong. I'm running ROS nodes and `zenoh_bridge_ros2dds` in a docker container. The OS of host B is Ubuntu 20.04, but the docker containers on both hosts are Ubuntu 22.04 based.

> Since the issue might not be reproduced easily, it would be great if you could provide the log in zenoh-bridge-ros2dds while the issue happens (Please add `RUST_LOG=z=debug` while running the bridge)

It is difficult to share the whole logs as there can be confidential information, so let me extract general logs lines and lines include `/diagnostics`. I saved the raw log files, so I can search the logs if you want to see whether a specific message exists or not.

## Logs of `zenoh_bridge_ros2dds` on host A

Logs right after `zenoh_bridge_ros2dds` is launched. At this moment, `zenoh_bridge_ros2dds` doesn't forward messages on `/diagnostics` topic, even there's a subscriber (`diagnostic_aggregator` node) to the topic on host A.

```
2024-11-20T11:18:29+09:00 - [INFO] [zenoh_bridge_ros2dds-59]: process started with pid [916]
2024-11-20T02:18:29.525232Z DEBUG tokio-runtime-worker ThreadId(07) zenoh_plugin_ros2dds: Node /navigation_planner declares Publisher /rosout: rcl_interfaces/msg/Log - Denied per config
2024-11-20T02:18:30.176945Z DEBUG                 rx-0 ThreadId(14) zenoh::net::routing::dispatcher::token: Face{2, 72be8e633d5b028b68afb317e32d29e0} Declare token 0 (@/72be8e633d5b028b68afb317e32d29e0/@ros2_lv/MP/diagnostics/diagnostic_msgs§msg§DiagnosticArray/:1::)
2024-11-20T02:18:30.176734Z DEBUG                 rx-0 ThreadId(14) zenoh::net::routing::dispatcher::resource: Register resource @/72be8e633d5b028b68afb317e32d29e0/@ros2_lv/MP/diagnostics/diagnostic_msgs§msg§DiagnosticArray/:1::
2024-11-20T02:18:32.273227Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS Participant 0110268af382e12b94d2b33c000001c1)
2024-11-20T02:18:32.448578Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS subscription 0110268af382e12b94d2b33c00000504 from Participant 0110268af382e12b94d2b33c000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
2024-11-20T02:18:32.461705Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 0110268af382e12b94d2b33c00000603 from Participant 0110268af382e12b94d2b33c000001c1 on rt/rosout with type rcl_interfaces::msg::dds_::Log_ (keyless: true)
2024-11-20T02:18:32.461659Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 0110268af382e12b94d2b33c00000403 from Participant 0110268af382e12b94d2b33c000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
2024-11-20T02:18:32.448821Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS subscription 0110268af382e12b94d2b33c00001404 from Participant 0110268af382e12b94d2b33c000001c1 on rt/parameter_events with type rcl_interfaces::msg::dds_::ParameterEvent_ (keyless: true)
2024-11-20T02:18:32.468893Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 0110268af382e12b94d2b33c00001303 from Participant 0110268af382e12b94d2b33c000001c1 on rt/parameter_events with type rcl_interfaces::msg::dds_::ParameterEvent_ (keyless: true)
2024-11-20T02:18:34.232917Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS Participant 0110f0e323f4cd2a4cdbed11000001c1)
2024-11-20T02:18:34.300041Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 0110f0e323f4cd2a4cdbed1100000603 from Participant 0110f0e323f4cd2a4cdbed11000001c1 on rt/rosout with type rcl_interfaces::msg::dds_::Log_ (keyless: true)
2024-11-20T02:18:34.299992Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 0110f0e323f4cd2a4cdbed1100000403 from Participant 0110f0e323f4cd2a4cdbed11000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
2024-11-20T02:18:34.301096Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS subscription 0110f0e323f4cd2a4cdbed1100000504 from Participant 0110f0e323f4cd2a4cdbed11000001c1 on ros_discovery_info with type rmw_dds_common::msg::dds_::ParticipantEntitiesInfo_ (keyless: true)
2024-11-20T02:18:34.301078Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS publication 0110f0e323f4cd2a4cdbed1100001303 from Participant 0110f0e323f4cd2a4cdbed11000001c1 on rt/parameter_events with type rcl_interfaces::msg::dds_::ParameterEvent_ (keyless: true)
2024-11-20T02:18:34.307710Z DEBUG ThreadId(12) zenoh_plugin_ros2dds::dds_discovery: Discovered DDS subscription 0110f0e323f4cd2a4cdbed1100001404 from Participant 0110f0e323f4cd2a4cdbed11000001c1 on rt/parameter_events with type rcl_interfaces::msg::dds_::ParameterEvent_ (keyless: true)
2024-11-20T02:18:34.778643Z DEBUG tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::discovered_entities: ROS Node /tf_publisher declares a new Writer on rt/tf_static
[... cut ...]
2024-11-20T02:28:08.639414Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /_ros2cli_daemon_10_85b832c30aca44568a066080a5f4acf3 declares Publisher /parameter_events: rcl_interfaces/msg/ParameterEvent - Allowed
2024-11-20T02:28:08.638812Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Node /_ros2cli_daemon_10_85b832c30aca44568a066080a5f4acf3 declares Publisher /rosout: rcl_interfaces/msg/Log - Allowed
2024-11-20T02:28:40.309089Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds: Remote bridge a87c89febb0205556e67db38a53489b0 announces Subscriber diagnostics
```

JEnoch, 2024-11-25:
Looking at your logs of `zenoh_bridge_ros2dds` on host A:
- In the first part of the log, the lines 3 and 4 from `rx-0 ThreadId(14)` show that the bridge is receiving the declaration from the remote bridge (host B) of a DDS Publisher (`/MP/` for "Message Publisher" in the key expr).
  A local DDS Publisher is also discovered at line 24. 
  But there is no evidence of discovery of a local DDS Subscriber on `/diagnostics` topic.
- In the second part of the log, we see the bridge is now well discovering the DDS Subscriber on `/diagnostic` from Node `/_ros2cli_10219` (i.e. the `ros2 topic echo` command).

So it seems the issue lies in the DDS discovery between the bridge on host A and your `diagnostic_aggregator` Node.
To investigate further, you should activate the CycloneDDS logging for `discovery` category on both bridge and `diagnostic_aggregator`  Node.
You can do this defining this environment variable for each:
`CYCLONEDDS_URI='<Tracing><Category>discovery</><Verbosity>info</><Out>stdout</></>'`

If the bridge is well receiving the SEDP message (DDS discovery message) for the `/diagnostics` topic, you should see such log:
```log
1732558656.302033 [0] dq.builtin: SEDP ST0 110f30f:d653004:71b94e83:1504 reliable volatile reader unnamed: (default).rt/diagnostics/diagnostic_msgs::msg::dds_::DiagnosticArray_ NEW (as udp/239.255.0.1:7401@1 udp/127.0.0.1:57630@1 ssm=0) QOS={user_data=0<>,topic_name="rt/diagnostics"   ...
```

If the Node is well announcing its Subscriber, you should see such log:
```log
1732558964.004424 [0]   42929409: new_reader(guid 11053f2:86813d58:5a9a2766:1504, (default).rt/diagnostics/diagnostic_msgs::msg::dds_::DiagnosticArray_)
1732558964.004434 [0]   42929409: READER 11053f2:86813d58:5a9a2766:1504 QOS={user_data=0<>,topic_name="rt/diagnostics" ...
```

yuma-m, 2024-11-29:
@JEnoch Thank you so much for helping me investigate this issue. I've collected the logs from CycloneDDS of both hosts when the issue happened.

From logs of host A, I found the SEDP message for `/diagnostics` topic.

```
1732862340.656221 [10] dq.builtin: SEDP ST0 1102aea:a6d03be3:8583004e:1604 reliable volatile reader unnamed: (default).rt/diagnostics/diagnostic_msgs::msg::dds_::DiagnosticArray_ p(open) NEW (as udp/239.255.0.1:9901@331 udp/192.168.10.3:52120@331 ssm=0) QOS={user_data=0<>,topic_name="rt/diagnostics",type_name="diagnostic_msgs::msg::dds_::DiagnosticArray_",topic_data=0<>,group_data=0<>,durability=0,deadline=9223372036854775807,latency_budget=0,liveliness=0:9223372036854775807,reliability=1:9223372036854775807,destination_order=0,history=0:1000,resource_limits=-1:-1:-1,ownership=0,presentation=0:0:0,partition={},time_based_filter=0,transport_priority=0,type_consistency_enforcement=1:11000,data_representation=1(0),adlink_entity_factory=1,adlink_reader_lifespan=0:9223372036854775807,adlink_reader_data_lifecycle=9223372036854775807:9223372036854775807,adlink_subscription_keys=0:{}}
```

However, I couldn't find logs of announcing Subscriber around the time when the `diagnostic_aggregator` node was launched.

When I ran `ros2 topic echo /diagnostics` manually after the issue occured, the following logs were recorded.

```
1732861261.740705 [10]       ros2: new_reader(guid 110a46c:9c1aa783:8aea6191:804, (default).rt/diagnostics/diagnostic_msgs::msg::dds_::DiagnosticArray_)
1732861261.740737 [10]       ros2: READER 110a46c:9c1aa783:8aea6191:804 QOS={user_data=0<>,topic_name="rt/diagnostics",type_name="diagnostic_msgs::msg::dds_::DiagnosticArray_",topic_data=0<>,group_data=0<>,durability=0,deadline=9223372036854775807,latency_budget=0,liveliness=0:9223372036854775807,reliability=1:9223372036854775807,destination_order=0,history=0:5,resource_limits=-1:-1:-1,ownership=0,presentation=0:0:0,partition={},time_based_filter=0,transport_priority=0,type_consistency_enforcement=1:11000,data_representation=1(0),adlink_entity_factory=1,adlink_reader_lifespan=0:9223372036854775807,adlink_reader_data_lifecycle=9223372036854775807:9223372036854775807,adlink_subscription_keys=0:{}}
```

What do you think from these results? Is this issue originated from CycloneDDS's discovery failure?

JEnoch, 2024-11-29:
Can you clarify which processes are showing those respective logs (bridge or which Node) ?

If the bridge is logging the reception of SEDP Reader message for `/diagnostic` it means the CycloneDDS discovery is successful. So I'm not sure what happens...

However I just merged a bug fix (#346) that might be an explanation to the issue you experience. Could please you try with the latest commit from `main` branch ?

If you still experience the issue, I think a would need the complete logs from both bridges running with such environment variables:
```log
CYCLONEDDS_URI='<Tracing><Category>discovery</><Verbosity>info</><Out>stdout</></>'
RUST_LOG=zenoh_plugin=trace
```
You can grep the resulting logs with `/diagnostics` only before providing them.

yuma-m, 2024-12-23:
Hi @JEnoch, thank you for your information and sorry for my late response.

> Can you clarify which processes are showing those respective logs (bridge or which Node) ?

The logs mentioned in my previous comment are from the Zenoh bridge on host A.

> Could please you try with the latest commit from main branch ?

I've updated the Zenoh bridge to version v1.0.4, which includes your fix, but I'm still encountering the same issue.

> If you still experience the issue, I think a would need the complete logs from both bridges running with such environment variables:

I tested it again with the environment variables you specified. Here are the logs from the Zenoh bridge on host A when the issue occurred. Could you kindly take a look again?

```
[zenoh_bridge_ros2dds-58] 1734920533.447165 [10] dq.builtin: SEDP ST0 1108af3:f273fab8:faf02587:1503 reliable volatile writer unnamed: (default).rt/diagnostics/diagnostic_msgs::msg::dds_::DiagnosticArray_ NEW (as udp/239.255.0.1:9901@223 udp/192.168.10.3:56729@223 ssm=0) QOS={user_data=0<>,topic_name="rt/diagnostics",type_name="diagnostic_msgs::msg::dds_::DiagnosticArray_",topic_data=0<>,group_data=0<>,durability=0,durability_service=0:0:1:-1:-1:-1,deadline=9223372036854775807,latency_budget=0,liveliness=0:9223372036854775807,reliability=1:9223372036854775807,lifespan=9223372036854775807,destination_order=0,history=0:1,resource_limits=-1:-1:-1,ownership=0,ownership_strength=0,presentation=0:0:0,partition={},transport_priority=0,data_representation=1(0),adlink_entity_factory=1,adlink_writer_data_lifecycle=0}
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:13.447203Z[0m [34mDEBUG[0m ThreadId(12) [2mzenoh_plugin_ros2dds::dds_discovery[0m[2m:[0m Discovered DDS publication 01108af3f273fab8faf0258700001503 from Participant 01108af3f273fab8faf02587000001c1 on rt/diagnostics with type diagnostic_msgs::msg::dds_::DiagnosticArray_ (keyless: true)
[zenoh_bridge_ros2dds-58] 1734920533.447232 [10] dq.builtin: match_proxy_writer_with_readers(pwr 1108af3:f273fab8:faf02587:1503) scanning all rds of topic rt/diagnostics
[zenoh_bridge_ros2dds-58] 1734920538.351882 [10] dq.builtin: SEDP ST0 1109b5d:7604c7d:c8097f19:1503 reliable volatile writer unnamed: (default).rt/diagnostics/diagnostic_msgs::msg::dds_::DiagnosticArray_ NEW (as udp/239.255.0.1:9901@223 udp/192.168.10.3:47372@223 ssm=0) QOS={user_data=0<>,topic_name="rt/diagnostics",type_name="diagnostic_msgs::msg::dds_::DiagnosticArray_",topic_data=0<>,group_data=0<>,durability=0,durability_service=0:0:1:-1:-1:-1,deadline=9223372036854775807,latency_budget=0,liveliness=0:9223372036854775807,reliability=1:9223372036854775807,lifespan=9223372036854775807,destination_order=0,history=0:1,resource_limits=-1:-1:-1,ownership=0,ownership_strength=0,presentation=0:0:0,partition={},transport_priority=0,data_representation=1(0),adlink_entity_factory=1,adlink_writer_data_lifecycle=0}
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:18.351920Z[0m [34mDEBUG[0m ThreadId(12) [2mzenoh_plugin_ros2dds::dds_discovery[0m[2m:[0m Discovered DDS publication 01109b5d07604c7dc8097f1900001503 from Participant 01109b5d07604c7dc8097f19000001c1 on rt/diagnostics with type diagnostic_msgs::msg::dds_::DiagnosticArray_ (keyless: true)
[zenoh_bridge_ros2dds-58] 1734920538.351954 [10] dq.builtin: match_proxy_writer_with_readers(pwr 1109b5d:7604c7d:c8097f19:1503) scanning all rds of topic rt/diagnostics
```

Additionally, I collected the logs from when the Zenoh bridge was able to discover a subscriber for the `/diagnostics` topic (when the issue did not occur).

```
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.964955Z[0m [35mTRACE[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::discovered_entities[0m[2m:[0m ROS Node /diagnostic_aggregator declares a Reader on rt/diagnostics
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.964958Z[0m [34mDEBUG[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::discovered_entities[0m[2m:[0m ROS Node /diagnostic_aggregator declares a new Reader on rt/diagnostics
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.964978Z[0m [35mTRACE[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::discovered_entities[0m[2m:[0m ROS Node /diagnostic_aggregator declares Writer on rt/diagnostics_toplevel_state
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.964980Z[0m [34mDEBUG[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::discovered_entities[0m[2m:[0m ROS Node /diagnostic_aggregator declares a new Writer on rt/diagnostics_toplevel_state
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.968858Z[0m [32m INFO[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds[0m[2m:[0m Node /diagnostic_aggregator declares Subscriber /diagnostics: diagnostic_msgs/msg/DiagnosticArray - Allowed
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.968873Z[0m [34mDEBUG[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::route_subscriber[0m[2m:[0m Route Subscriber (diagnostics -> /diagnostics): creation with type diagnostic_msgs/msg/DiagnosticArray (transient_local:false)
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.968878Z[0m [34mDEBUG[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::route_subscriber[0m[2m:[0m Route Subscriber (diagnostics -> /diagnostics): create Writer with Qos { user_data: None, topic_data: None, group_data: None, durability: None, durability_service: None, presentation: None, deadline: None, latency_bud [... cut ...]
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.969038Z[0m [34mDEBUG[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::route_subscriber[0m[2m:[0m Route Subscriber (Zenoh:diagnostics -> ROS:/diagnostics) now serving local nodes {"/diagnostic_aggregator"}
[zenoh_bridge_ros2dds-58] [2m2024-12-23T02:22:25.969042Z[0m [34mDEBUG[0m tokio-runtime-worker ThreadId(08) [2mzenoh_plugin_ros2dds::route_subscriber[0m[2m:[0m Route Subscriber (Zenoh:diagnostics -> ROS:/diagnostics) activate
```