eclipse-zenoh/zenoh-plugin-ros2dds #339: Release into the buildfarm

luca-della-vedova, 2024-11-20:
### Describe the feature

This package is currently not released to the buildfarm (and the `zenoh-plugin-dds` is released but only at version 0.5 and has been failing for a long time).
It would be great if they can be released since ROS 2 users are the main target audience of these packages and to be installed they currently require either compilation from source or installing from the eclipse repo.

By contrast, if they were released in the buildfarm ROS users would just need to add them to their `package.xml` and installation would be automatically handled by `rosdep`.

I'm not sure whether there is any major blocker, I'm happy to help if that is the case

JEnoch, 2024-11-22:
I don't believe there are any major obstacles aside from the time required to implement and maintain it...  😅
So, your offer to help is very welcome!

Now that we have a clean CI and release train, with more frequent releases (see the latest 1.0.x [released](https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/releases) in the last few weeks), I think it would be beneficial to include the release into the build farm as part of our release workflow here:
https://github.com/eclipse-zenoh/zenoh-plugin-ros2dds/blob/main/.github/workflows/release.yml

Could you provide some guidance on how to achieve this?

Timple, 2025-01-09:
I would be interested in this as well.

Do you still need any guidance @JEnoch? As `rmw_zenoh` just went through the same process?

emersonknapp, 2025-03-17:
It isn't too hard to do this manually either, depending on how often you're cutting new releases. From my experience, adoption of even extremely useful packages is highly limited by `apt-get install ros-humble-xxxxxx` availability.