eclipse-zenoh/zenoh-plugin-ros2dds #382: [Bug] Service Client destroyed when no Service Server exists causes complete stop

Morgantiz, 2025-01-21:
### Describe the bug

I'm using zenoh bridge to communicate between two PCs with [CycloneDDS](https://github.com/eclipse-cyclonedds/cyclonedds) as RMW with HUMBLE ROS2 distro. The two PCs are connected via WiFi and they are both running zenoh_bridge_ros2dds one with 'router' mode and the other one with 'client' mode. I'm running the zenoh bridge on a Docker container with base image ros-humble-ros-base and zenoh_bridge_ros2dds installed from source following the README guidelines.

### To reproduce

Communication test: 

1. Start `zenoh_bridge_ros2dds` on router PC
2. Start `zenoh_bridge_ros2dds` on client PC
3. Start talker `ros2 run demo_nodes_cpp talker` on router PC
4. Start listener `ros2 run demo_nodes_cpp listener` on client PC 

When listener started listening the talker i followed these steps:

1. Run the service client `ros2 run demo_nodes_cpp add_two_ints_client`
2. Stop the add_two_ints_client node (for example with Keyboard interruption `^C`)

Now the listener does not receives anymore the messages coming from the talker.

Zenoh router log output: 

```
2025-01-21T13:29:29.693912Z  INFO main ThreadId(01) zenoh_bridge_ros2dds: zenoh-bridge-ros2dds v1.1.1
2025-01-21T13:29:29.694910Z  INFO main ThreadId(01) zenoh_bridge_ros2dds: Zenoh Config(Config { id: 7548557fa139b3481bd072935a16100c, metadata: Null, mode: Some(Router), connect: ConnectConfig { timeout_ms: None, endpoints: Unique([]), exit_on_failure: None, retry: None }, listen: ListenConfig { timeout_ms: None, endpoints: Dependent(ModeValues { router: Some([tcp/[::]:7447]), peer: Some([tcp/[::]:0]), client: None }), exit_on_failure: None, retry: None }, open: OpenConf { return_conditions: ReturnConditionsConf { connect_scouted: None, declares: None } }, scouting: ScoutingConf { timeout: None, delay: None, multicast: ScoutingMulticastConf { enabled: None, address: None, interface: None, ttl: None, autoconnect: None, listen: None }, gossip: GossipConf { enabled: None, multihop: None, target: None, autoconnect: None } }, timestamping: TimestampingConf { enabled: Some(Unique(true)), drop_future_timestamp: None }, queries_default_timeout: None, routing: RoutingConf { router: RouterRoutingConf { peers_failover_brokering: None }, peer: PeerRoutingConf { mode: None }, interests: InterestsConf { timeout: None } }, aggregation: AggregationConf { subscribers: [], publishers: [] }, qos: QoSConfig { publication: PublisherQoSConfList([]) }, transport: TransportConf { unicast: TransportUnicastConf { open_timeout: 10000, accept_timeout: 10000, accept_pending: 100, max_sessions: 1000, max_links: 1, lowlatency: false, qos: QoSUnicastConf { enabled: true }, compression: CompressionUnicastConf { enabled: false } }, multicast: TransportMulticastConf { join_interval: Some(2500), max_sessions: Some(1000), qos: QoSMulticastConf { enabled: false }, compression: CompressionMulticastConf { enabled: false } }, link: TransportLinkConf { protocols: None, tx: LinkTxConf { sequence_number_resolution: U32, lease: 10000, keep_alive: 4, batch_size: 65535, queue: QueueConf { size: QueueSizeConf { control: 1, real_time: 1, interactive_high: 1, interactive_low: 1, data_high: 2, data: 4, data_low: 2, background: 1 }, congestion_control: CongestionControlConf { drop: CongestionControlDropConf { wait_before_drop: 1000, max_wait_before_drop_fragments: 50000 }, block: CongestionControlBlockConf { wait_before_close: 5000000 } }, batching: BatchingConf { enabled: true, time_limit: 1 } }, threads: 3 }, rx: LinkRxConf { buffer_size: 65535, max_message_size: 1073741824 }, tls: TLSConf { root_ca_certificate: None, listen_private_key: None, listen_certificate: None, enable_mtls: None, connect_private_key: None, connect_certificate: None, verify_name_on_connect: None, close_link_on_expiration: None, so_sndbuf: None, so_rcvbuf: None, root_ca_certificate_base64: None, listen_private_key_base64: None, listen_certificate_base64: None, connect_private_key_base64: None, connect_certificate_base64: None }, tcp: TcpConf { so_sndbuf: None, so_rcvbuf: None }, unixpipe: UnixPipeConf { file_access_mask: None } }, shared_memory: ShmConf { enabled: true }, auth: AuthConf { usrpwd: UsrPwdConf { user: None, password: None, dictionary_file: None }, pubkey: PubKeyConf { public_key_pem: None, private_key_pem: None, public_key_file: None, private_key_file: None, key_size: None, known_keys_file: None } } }, adminspace: AdminSpaceConf { enabled: true, permissions: PermissionsConf { read: true, write: false } }, downsampling: [], access_control: AclConfig { enabled: false, default_permission: Deny, rules: None, subjects: None, policies: None }, plugins_loading: PluginsLoading { enabled: true, search_dirs: LibSearchDirs([Spec(LibSearchSpec { kind: CurrentExeParent, value: None }), Path("."), Path("~/.zenoh/lib"), Path("/opt/homebrew/lib"), Path("/usr/local/lib"), Path("/usr/lib")]) }, plugins: Object {"ros2dds": Object {"domain": Number(73), "ros_localhost_only": Bool(true)}} })
2025-01-21T13:29:29.695039Z  INFO main ThreadId(01) zenoh::net::runtime: Using ZID: 7548557fa139b3481bd072935a16100c
2025-01-21T13:29:29.695377Z  INFO main ThreadId(01) zenoh::api::loader: Starting required plugin "ros2dds"
2025-01-21T13:29:29.695674Z  INFO main ThreadId(01) zenoh::api::loader: Successfully started plugin ros2dds from "__static_lib__"
2025-01-21T13:29:29.695688Z  INFO main ThreadId(01) zenoh::api::loader: Finished loading plugins
2025-01-21T13:29:29.695696Z  INFO tokio-runtime-worker ThreadId(13) zenoh_plugin_ros2dds: ROS2 plugin Config { namespace: "/", nodename: "zenoh_bridge_ros2dds", domain: 73, ros_localhost_only: true, ros_automatic_discovery_range: None, ros_static_peers: None, allowance: None, pub_max_frequencies: [], transient_local_cache_multiplier: 10, queries_timeout: Some(QueriesTimeouts { default: 5.0, transient_local_subscribers: [], services: [], actions: Some(ActionsTimeouts { send_goal: [], cancel_goal: [], get_result: [(Regex(".*"), 300.0)] }) }), reliable_routes_blocking: true, pub_priorities: [], work_thread_num: 2, max_block_thread_num: 50, __required__: None, __path__: None }
... 

2025-01-21T13:29:44.593203Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::discovered_entities: Discovered ROS Node /talker
2025-01-21T13:29:44.593311Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds: Node /talker declares Subscriber /parameter_events: rcl_interfaces/msg/ParameterEvent - Allowed
2025-01-21T13:29:44.593462Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::routes_mgr: Route Subscriber (Zenoh:parameter_events -> ROS:/parameter_events) created
2025-01-21T13:29:44.593778Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds: Node /talker declares Service Server /talker/get_parameters: rcl_interfaces/srv/GetParameters - Allowed
2025-01-21T13:29:44.593957Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::routes_mgr: Route Service Server (ROS:/talker/get_parameters <-> Zenoh:talker/get_parameters) created
2025-01-21T13:29:44.594065Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds: Node /talker declares Service Server /talker/list_parameters: rcl_interfaces/srv/ListParameters - Allowed
2025-01-21T13:29:44.594201Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::routes_mgr: Route Service Server (ROS:/talker/list_parameters <-> Zenoh:talker/list_parameters) created
2025-01-21T13:29:44.594288Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds: Node /talker declares Publisher /rosout: rcl_interfaces/msg/Log - Allowed
2025-01-21T13:29:44.594605Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::routes_mgr: Route Publisher (ROS:/rosout -> Zenoh:rosout) created
2025-01-21T13:29:44.594733Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds: Node /talker declares Service Server /talker/get_parameter_types: rcl_interfaces/srv/GetParameterTypes - Allowed
2025-01-21T13:29:44.594893Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::routes_mgr: Route Service Server (ROS:/talker/get_parameter_types <-> Zenoh:talker/get_parameter_types) created
2025-01-21T13:29:44.594972Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds: Node /talker declares Publisher /chatter: std_msgs/msg/String - Allowed
2025-01-21T13:29:44.595007Z  INFO tokio-runtime-worker ThreadId(06) zenoh_plugin_ros2dds::routes_mgr: Route Publisher (ROS:/chatter -> Zenoh:chatter) created
[... cut ...]
2025-01-21T13:30:03.866437Z  INFO tokio-runtime-worker ThreadId(05) zenoh_plugin_ros2dds: Node /add_two_ints_client undeclares Service Client /add_two_ints: example_interfaces/srv/AddTwoInts - Allowed
2025-01-21T13:30:03.866483Z  INFO tokio-runtime-worker ThreadId(05) zenoh_plugin_ros2dds::routes_mgr: Route Service Client (ROS:/add_two_ints <-> Zenoh:add_two_ints) removed
2025-01-21T13:30:03.958355Z  INFO tokio-runtime-worker ThreadId(22) zenoh_plugin_ros2dds::discovered_entities: Undiscovered ROS Node /add_two_ints_client
```

### System info

Distributor ID: Ubuntu
Description:    Ubuntu 22.04.5 LTS
Release:        22.04
Codename:       jammy

ROS DISTRO: Humble 
ZENOH TAG: release/1.1.1

RPRKLR, 2025-01-27:
I experience the same issue, tried with `rmw_fastrtps_cpp` and `rmw_cyclonedds_cpp`.  
Distributor ID: Ubuntu
Description: Ubuntu 24.04.1 LTS
Release: 24.04
Codename: noble

ROS DISTRO: Jazzy
ZENOH TAG: release/1.1.1

czarekk215, 2025-02-28:
I experienced similar issue, but my issue is connected also with storing data in influx via zenoh from ros2. When I add storage via rest api to store data from topic X to zenoh, after few service calls to add_two_ints server it stops recording data to influx and it causes complete stop.

Schema of my test case:

[image]

Distributor ID: Ubuntu
Description: Ubuntu 22.04.4

ROS DISTRO: Humble
ZENOH TAG: release/1.2.1

cesurs, 2025-03-20:
I have the exact same problem. To localize the problem the only thing I am doing in my node is to create a service client. When I shutdown the node 2 devices can no longer communicate until I restart the bridge on the publisher side. I am using bridge through docker and ROS2 Humble as well.

adamkrawczyk, 2025-03-24:
@JEnoch Do you have this issue on your radar?

edu-rossrobotics, 2025-04-29:
Hi, is there any prevision on when this may be addressed? I'm happy to contribute to it as this is a really important thing for us to sort out but I'm not really familiar with zenoh so I' unsure where the issue may be coming from.

JEnoch, 2025-09-04:
Hi all,
I finally found time to dig into this, and found the (weird) cause.
Fix is coming (#533).

commit f26cec3b (2025-09-05): Set all MatchingListener as background (fix #382) (#533)