eclipse-zenoh/zenoh-plugin-ros2dds #404: [Bug] re-subscription impossible while peer subscribes its own topic

rayferric, 2025-02-14:
### Describe the bug

As per the title. The issue makes it impossible to RViz internal topics used by the robot on a dev-laptop.

### To reproduce

Setup Cyclone DDS with ROS_LOCALHOST_ONLY=1.

Terminal #1:
```
ROS_DOMAIN_ID=1 zenoh_bridge_ros2dds-1.2.1 -l tcp/localhost:7000
```
Terminal #2:
```
ROS_DOMAIN_ID=2 zenoh_bridge_ros2dds-1.2.1 -e tcp/localhost:7000
```
Terminal #3:
```
ROS_DOMAIN_ID=1 ros2 topic pub /foobar std_msgs/msg/Bool "{data: true}"
```
Terminal #4:
```
ROS_DOMAIN_ID=2 ros2 topic echo /foobar
```
Terminal $5:
```
ROS_DOMAIN_ID=1 ros2 topic echo /foobar
```
Terminal #4:
```
CTRL+C
ROS_DOMAIN_ID=2 ros2 topic echo /foobar
```
The messages will not flow until Terminal #4 resubscribes while Terminal #5 is unsubscribed.

### System info

- Platform: Ubuntu Jammy/Humble (distrobox on Arch Linux x86_64 host)
- CPU: AMD Ryzen 7 5800H (16) @ 3.0GHz
- Zenoh Version: Release 1.2.1

rayferric, 2025-02-14:
Might be related to #368

rayferric, 2025-02-21:
I can't believe that more people aren't reporting this problem. It is really easy to encounter which makes it important to be resolved ASAP. It is straightforward to reproduce, so can anybody confirm that they see it?

breqdev, 2025-04-04:
Hey, we can confirm that this is affecting us as well! We've had repeated problems for the past year where topics stop flowing until the Zenoh bridge is restarted on either end, but haven't been able to narrow it to a minimal reproducible example like this. We can reproduce your example commands and believe this is the same issue we're experiencing at a larger scale.

guni9191, 2025-07-01:
Can't believe this still happens. Using 22.04 ubuntu, humble. Is this solved? maybe in 24.04?