eclipse-zenoh/zenoh-plugin-ros2dds #413: [Bug] Not able to run ros2 actions due to zenoh query timeout

Anisha-thakkar, 2025-02-27:
### Describe the bug

I am using zenoh-bridge-ros2dds for connecting my robot and server and communicate between them. I am running zenoh-bridge in router mode in the server and client mode on robot with config allowing all the topics and actions. After the connection is successfully done I am able to give cmd_vel to robot and other topics from server to run the robot and when trying run the action /navigate_to_pose, it throws timeout error, I am using latest version of zenoh-bridge 1.2.1, which has timeout changes for get_result 300ms, also I am providing config from robot side which explicitly defines timeout, stil while running action it throws (5s) timeout. I am unsure of from where it is coming from. I have tried all the version from 0.11.0, 1.0.0, 1.1.0 and 1.1.1, nothing worked for me. Are the action or services are broken still? I am attaching here the config I am using client (robot side) and video of the issue.

[client_config.zip](https://github.com/user-attachments/files/19003966/client_config.zip)

### To reproduce

1. zenoh-bridge-ros2dds -e tcp/<clientip>:7447  (in server)
2. zenoh-bridge-ros2dds -c client_config.json5  (in client)
3. ros2 action send_goal /<namespace>/navigate_to_pose nav2_msgs/action/NavigateToPose "pose: {header: {frame_id: map}, pose: {position: {x: 0.0, y: 0.0, z: 0.0}, orientation:{x: 0.0, y: 0.0, z: 0, w: 1.0000000}}}"

### System info

Server: Ubuntu'22
ROS2 Humble
Robot(client:) Ubuntu'22
ROS2 Humble

Anisha-thakkar, 2025-03-05:
I have tried to debug many ways but it is still the same. Through debugging I found one conclusion is when I am using /namespace for while connecting zenoh bridges this issue occurs, in case of no namespace It works find. What might be the root cause of this issue and solution for the same as I will need to use namespace because of I am working with multiple robots at a time.

tompe17, 2025-03-13:
I am seeing this also trying to run a Husky robot with rviz on a ground station.  Are there any workaround or more information about this?

EndlessLoops, 2025-11-14:
Are you fix the problem？