eclipse-zenoh/zenoh-plugin-ros2dds #435: [Bug] Freezes and does not recover - no error

stngrz, 2025-03-24:
### Describe the bug

My app has ros2 dds topics communicating locally (~90 topics @ max) on the same host (A) and communicating with a cloud container, the container has about 10 topics (continuous), host (B).

While in a steady state zenoh-bridge-ros2dds sends messages cleanly between hosts A to B with no issue. During the execution of my app I transition to various states/modes where the volume of topics can fluctuate greatly or not at all. 

About 90% of these transitions (either many topic add/remove or none) create a mode where, on host A, the router freezes. I define freeze by not exiting, showing no errors - it just hangs. I don't see any corresponding events in logs on host B - as if B expects A to be connected and healthy. Otherwise it performs nominally - the issue is just during these stage/mode transitions. When i kill a docker running many topics i see this error as well (aka not using the apps change stage/mode feature)

I've tried to review rosout to see if there is a difference in messages sent vs received on A and B but found nothing promising in the few samples I collected. I hoped it would reveal a problematic message being sent by my app on A- but nothing.

I expect that between app stages/mode transitions the router continues to send and receive data or that it provide errors when its not sending/receiving data.

This occurs with and with out using custom zenoh json5 configs.

in the attached debug logs I've removed the customer name and replaced it with 'my-company'

[debug_start-wait5s-freeze-wait5s.txt](https://github.com/user-attachments/files/19439340/debug_start-wait5s-freeze-wait5s.txt)
[debug_start-freeze-wait10s.txt](https://github.com/user-attachments/files/19439341/debug_start-freeze-wait10s.txt)

### To reproduce

1. Start zenoh router
2. add/remove topics / kill docker with many topics / stage change (sorry this is ill defined)
3. Router log freezes with no crash or log output

### System info

- Platform: linux/aarch64 
Distributor ID: Ubuntu
Description:    Ubuntu 22.04.4 LTS
Release:        22.04
Codename:       jammy

- CPU:
Architecture:            aarch64
  CPU op-mode(s):        32-bit, 64-bit
  Byte Order:            Little Endian
CPU(s):                  12
  On-line CPU(s) list:   0-11
Vendor ID:               ARM
  Model name:            Cortex-A78AE

- Zenoh: 
zenoh-bridge-ros2dds v1.3.0 (also occurred on 1.21.1)

stngrz, 2025-03-25:
I ran another test
2 local hosts(A,B) to 1 cloud container (C)
 
host A triggers a freeze
both A and B see the same freeze - via ros2 topic echo of topic from C
if reset on C then B resumes but not A - must reset on A for A to resume
if reset on A then both A and B resume
did not try reset on B when frozen

stngrz, 2025-04-01:
Another data point, expecting that Local Host does not see zenoh pub if Cloud Host zenoh is not subbed

invalid state found
Cloud Host
pub = simulation
sub = none
Local Host
pub = zenoh_bridge
sub = perception_manager

edu-rossrobotics, 2025-04-29:
Hi, did you manage to fix it? I have a similar issue where on my robot I launch a bunch of different nodes, when the first one comes up, everything is working but when I launch the node that controls the robot movements (it basically involves launching ros2control packages) I got the same issue:

```
2025-04-29T09:01:00.093590Z  INFO tokio-runtime-worker ThreadId(03) zenoh_plugin_ros2dds::routes_mgr: Route Service Client (ROS:/controller_manager/list_controllers <-> Zenoh:controller_manager/list_controllers) removed
2025-04-29T09:01:00.158524Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds::discovered_entities: Undiscovered ROS Node /spawner_joint_state_broadcaster
2025-04-29T09:01:01.200437Z  INFO tokio-runtime-worker ThreadId(02) zenoh_plugin_ros2dds::discovered_entities: Undiscovered ROS Node /spawner_mk4_base_controller
2025-04-29T09:01:02.268224Z  INFO tokio-runtime-worker ThreadId(04) zenoh_plugin_ros2dds::discovered_entities: Undiscovered ROS Node /controller_manager
2025-04-29T09:01:05.381858Z  WARN                net-0 ThreadId(06) zenoh::net::routing::dispatcher::queries: Face{1, ebdefef004106e20966a87c2cb6f460b}:79 Didn't receive final reply for query Face{2, afa6a071508ccc64ac0aa86eaf70f153}:79: Timeout(5s)!
```

And all the communication stops working. But, just restarting the zenoh bit makes it work so I assume the issue is during the launch process.